cyberman54/ESP32-Paxcounter #1: TTGO v1

Octoate, 2018-03-19:
Hi,
I am using a TTGO v1 board, but I ma unable to use it with the current version of the Paxcounter, because it seems not to be able to use WiFi. Here is the log of it:

> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:812
> load:0x40078000,len:0
> load:0x40078000,len:11404
> entry 0x40078a28
> [I][main.cpp:327] setup(): Starting PAXCNT 1.2.0
> [I][main.cpp:341] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
> [I][configmanager.cpp:130] loadConfig(): Reading settings from NVS
> [I][configmanager.cpp:41] open_storage(): Opening NVS
> [I][configmanager.cpp:46] open_storage(): Done
> [I][configmanager.cpp:143] loadConfig(): NVRAM settings version = 1.2.0
> [I][configmanager.cpp:159] loadConfig(): lorasf = 9
> [I][configmanager.cpp:167] loadConfig(): txpower = 15
> [I][configmanager.cpp:175] loadConfig(): adrmode = 1
> [I][configmanager.cpp:183] loadConfig(): screensaver = 0
> [I][configmanager.cpp:191] loadConfig(): screenon = 1
> [I][configmanager.cpp:199] loadConfig(): countermode = 0
> [I][configmanager.cpp:207] loadConfig(): wifiscancycle = 120
> [I][configmanager.cpp:215] loadConfig(): wifichancycle = 50
> [I][configmanager.cpp:223] loadConfig(): blescancycle = 30
> [I][configmanager.cpp:231] loadConfig(): BLEscanmode = 0
> [I][configmanager.cpp:239] loadConfig(): rssilimit = 0
> [I][configmanager.cpp:246] loadConfig(): Done
> [I][lorawan.cpp:56] printKey(): DevEUI: xxxxx
> [I][lorawan.cpp:56] printKey(): AppEUI: yyyyy
> [I][lorawan.cpp:56] printKey(): AppKey: zzzzz
> I (8750) wifi: wifi firmware version: 403db1d
> I (8750) wifi: config NVS flash: enabled
> I (8750) wifi: config nano formating: disabled
> I (8755) wifi: Init dynamic tx buffer num: 32
> I (8756) wifi: Init data frame dynamic rx buffer num: 64
> I (8756) wifi: Init management frame dynamic rx buffer num: 64
> I (8759) wifi: wifi driver task: 3ffdbef4, prio:23, stack:4096
> I (8765) wifi: Init static rx buffer num: 10
> I (8769) wifi: Init dynamic rx buffer num: 0
> I (8773) wifi: wifi power manager task: 0x3ffe1054 prio: 21 stack: 2560
> I (17641) wifi: set country: cc=EU schan=1 nchan=13 policy=0
> 
> W (17648) wifi: sta start fail
> 
> E (17650) tcpip_adapter: tcpip_adapter is not initialized!
> I (17673) wifi: mode : null
> abort() was called at PC 0x40189f37 on core 0
> 
> Backtrace: 0x4008af28:0x3ffd64a0 0x4008b027:0x3ffd64c0 0x40189f37:0x3ffd64e0 0x4012b3cd:0x3ffd6520 0x4012b671:0x3ffd6540 0x40129c41:0x3ffd6560
> 
> Rebooting...
> ets Jun  8 2016 00:22:57

Any idea, what's going on here?

Btw, I also added a TTGO v1 configuration file and changed the platformio.ini to support it. I have attached them to this posting.

[platformio.ini.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/1826620/platformio.ini.txt)
[ttgov1.h.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/1826625/ttgov1.h.txt)

cyberman54, 2018-03-19:
I never tested the code on a TTGOv1 board. I did not buy one, because several issues were reported on this board in TTN forum. The board is reported to feature a SX1278. Not sure if this makes any difference to SX1272 and SX1276 for which the code was tested.

But the reported error seems to point to a wifi, not a LoRa related problem.

For further diagnostics try backtrace with the reported value of the program counter (PC) to find out at which part of the code the module resets. 

An abort could be intenially caused by the configASSERT statement in the wifi_sniffer_loop() in main.cpp:

> //WiFi Sniffer Task
> void wifi_sniffer_loop(void * pvParameters) {
> 
>     configASSERT( ( ( uint32_t ) pvParameters ) == 1 ); // FreeRTOS check

Try to comment out the line with configASSERT and try again, checking the logs.

cyberman54, 2018-03-19:
For reference, a log on a TTGOv2 board looks like this:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
[I][main.cpp:327] setup(): Starting PAXCNT 1.2.0
[I][main.cpp:341] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
[I][configmanager.cpp:130] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:41] open_storage(): Opening NVS
[I][configmanager.cpp:46] open_storage(): Done
[I][configmanager.cpp:143] loadConfig(): NVRAM settings version = 1.2.0
[I][configmanager.cpp:159] loadConfig(): lorasf = 9
[I][configmanager.cpp:167] loadConfig(): txpower = 15
[I][configmanager.cpp:175] loadConfig(): adrmode = 1
[I][configmanager.cpp:183] loadConfig(): screensaver = 0
[I][configmanager.cpp:191] loadConfig(): screenon = 1
[I][configmanager.cpp:199] loadConfig(): countermode = 0
[I][configmanager.cpp:207] loadConfig(): wifiscancycle = 120
[I][configmanager.cpp:215] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:223] loadConfig(): blescancycle = 30
[I][configmanager.cpp:231] loadConfig(): BLEscanmode = 1
[I][configmanager.cpp:239] loadConfig(): rssilimit = 0
[I][configmanager.cpp:246] loadConfig(): Done
[I][lorawan.cpp:56] printKey(): DevEUI: xxxxxxx
[I][lorawan.cpp:56] printKey(): AppEUI: xxxxxxx
[I][lorawan.cpp:56] printKey(): AppKey: xxxxxxx
I (8754) wifi: config NVS flash: enabled
I (8754) wifi: config nano formating: disabled
I (8764) wifi: Init dynamic tx buffer num: 32
I (8764) wifi: Init data frame dynamic rx buffer num: 64
I (8764) wifi: Init management frame dynamic rx buffer num: 64
I (8768) wifi: wifi driver task: 3ffdbfb0, prio:23, stack:4096
I (8773) wifi: Init static rx buffer num: 10
I (8777) wifi: Init dynamic rx buffer num: 0
I (8781) wifi: wifi power manager task: 0x3ffdfc28 prio: 21 stack: 2560
I (17649) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=0

W (17655) wifi: ap start fail

I (17680) wifi: mode : null
I (17681) wifi: ic_enable_sniffer
[I][main.cpp:390] setup(): Starting Lora task on core 1
[I][main.cpp:392] setup(): Starting Wifi task on core 0
[I][lorawan.cpp:89] do_send(): Packet queued
[I][wifisniffer.cpp:82] wifi_sniffer_packet_handler(): WiFi RSSI: -93
[I][wifisniffer.cpp:89] wifi_sniffer_packet_handler(): MAC counter:    1
```

Octoate, 2018-03-19:
I removed the "configASSERT" line, but it is still not working. I noticed the line at the end where our logs differ:
> "W (17648) wifi: sta start fail"
and yours
> "W (17655) wifi: ap start fail"
Even changing the "esp_wifi_set_mode" to "WIFI_MODE_AP" doesn't change anything on that behaviour.

Do you have any hints on how to backtrace to the PC?

cyberman54, 2018-03-20:
I'm not an expert on this and not into JTAG debugging. Maybe you find these links useful:
[ESP32 forum](https://www.esp32.com/viewtopic.php?t=3861)
[ESP32 API guide](http://esp-idf.readthedocs.io/en/latest/api-guides/core_dump.html)

The TTGOv1 uses sames ESP32 SoC as Heltec board, as far as i know. The wifi monitor mode is part of the ESP32 SoC, does not need peripherals. So i don't see a reason why it should not work with TTGOv1 board.

It may be possible that the Wifi firmware of ESP32 SoC differs. Check in boot log.
It may be possible that some Wifi settings are stored in NVRAM, if you used the same board for connecting to a Wifi network before. Try to delete all settings.

cyberman54, 2018-03-20:
Since the abort seems to happen at a point where Wifi on the ESP32 is activated: What power supply are you using? Sure, that it feeds the TTGOv1 with enough peak power?

You may try changing your power supply to a bigger  one.

Octoate, 2018-03-20:
Normally I use my Notebook to power the ESP32, but I also checked my active USB Hub - unfortunately nothing. Isn't the complete SoC firmware compiled and uploaded by PlatformIO? However, I will have a look if there is a possibility to clean up the NVRAM.

Octoate, 2018-03-20:
Ok... I finally got it working. I compared the Paxcounter setup code with the one I use when I tested the promiscuous mode of the ESP32 and found the following line:

> tcpip_adapter_init();

If I understand it correctly, the line needs to be called before the **esp_event_loop_init** and the code works after I put it right before it in the main.cpp file.

> ESP_LOGI(TAG, "Starting %s %s", PROGNAME, PROGVERSION);
>
> tcpip_adapter_init();
> // ESP Event Loop
> esp_event_loop_init(NULL, NULL);

Hmm... would be nice if somebody who owns the TTGO v1 checks if it works with it, too.

cyberman54, 2018-03-20:
Not sure if this is the root caus, because the code is running on the previously supported boards. Perhaps this is a side effect.

Besides that, a TCP/IP stack is not needed for the paxcounter app. Since it draws ressources (memory and power) i intentionally disabled it.

cyberman54, 2018-03-20:
I created new branch [TTGOv1-test](https://github.com/cyberman54/ESP32-Paxcounter/tree/TTGOv1-test/) and merged your edits there. Can you test with your board, please?

cyberman54, 2018-03-20:
Changed some settings in wifisniffer.cpp, Wifi NVRAM settings now disabled.
Please try again with this version 1.2.21

cyberman54, 2018-03-21:
TTGOv1 support now implemented in 1.3.0, Issue closed.
Please re-open if problem persists.

closed 2018-03-21 by cyberman54