cyberman54/ESP32-Paxcounter #2: hash & salt MACs before store them in RAM

cyberman54, 2018-03-20:
- scramble MACs before storing them in RAM, to improve privacy
- add hash & salt function for this purpose

cyberman54, 2018-03-22:
Lösungsidee: 

Zuerst wird die (zuvor ggf. Vendor-gefilterte) erfasste MAC-Adresse gekürzt, dann ein SALT-Wert angehängt, darauf eine Hashfunktion angewendet und das Ergebnis wieder so gekürzt, dass es wie eine reguläre MAC-Adresse aufgebaut ist. Das Ergebnis ist ein Identifikator. Temporär im RAM werden nur die Identifikatoren gehalten. Diese allein bilden die Zählgrundlage.

Der SALT-Wert wird bei Gerätereset oder in bestimmten (Mindest-)Intervallen neu generiert. Somit sind die Identifikatoren nach Gerätereset oder Ablauf des Intervalls unterschiedlich zu den zuvor erfassten. Eine Wiedererkennung von Geräten/Personen ist damit ohne weiteres Fingerprinting (das nicht Bestandteil des Codes ist) nicht möglich, selbst dann nicht, wenn die Identifkatoren-Daten irgendwo über längeren Zeitraum gespeichert vorlägen.

Der SALT-Wert muss eine Zufallszahl sein, die nicht persistent, d. h. nur im RAM, flüchtig gespeichert wird.

trlafleur, 2018-03-23:
This might be of help...

forum.arduino.cc/index.php?topic=394998.0

cyberman54, 2018-03-31:
implemented (v1.2.6).

commit 8c93f44c (2019-02-22): timemanager fixes #2
closed 2018-03-31 by cyberman54