cyberman54/ESP32-Paxcounter #6: Byte Swap

trlafleur, 2018-03-21:
You need to do it for both.. DEVEUI and APPEUI

cyberman54, 2018-03-21:
Done.
(Hopefully people with existing loraconf.h will not complain...)

closed 2018-03-21 by cyberman54

trlafleur, 2018-03-21:
I have your code running on a TTGO Ver 1 board

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

trlafleur, 2018-03-21:
I did get error of DR_SF11 and 12 undefined

// help function to assign LoRa datarates to spreadfactor values
void switch_lora (int sf, int tx) {
    if ( tx > 20 ) return;
    cfg.txpower = tx;
    switch (sf) {
        case 7: LMIC_setDrTxpow(DR_SF7,tx); cfg.lorasf=sf; break;
        case 8: LMIC_setDrTxpow(DR_SF8,tx); cfg.lorasf=sf; break;
        case 9: LMIC_setDrTxpow(DR_SF9,tx); cfg.lorasf=sf; break;
        case 10: LMIC_setDrTxpow(DR_SF10,tx); cfg.lorasf=sf; break;
//        case 11: LMIC_setDrTxpow(DR_SF11,tx); cfg.lorasf=sf; break;
//        case 12: LMIC_setDrTxpow(DR_SF12,tx); cfg.lorasf=sf; break;
        default: break;
    }
}

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

cyberman54, 2018-03-21:
try to add these lines in the top of rcommand.cpp:

 #include <lmic.h>
 #include <hal/hal.h>

Does it fix the problem?

cyberman54, 2018-03-21:
fixed it in rcommand.cpp (probably was a side effect bug from recent code sanitization)

trlafleur, 2018-03-22:
Your building a great framework for other TTN application...
You may want to consider building a default framework code base, that
allows application to be added with call backs...

As such, adding a time of day function with alarms that could trigger task
would be very useful, as you have FreeRTOS, it might make it easy to add a
library like this one below to build on... Device could make a request for
time of day, and/or one could send time to the device(s) from time to time
to keep RTC up-to-date using a linux time epic.

https://www.arduino.cc/en/Reference/RTC

adding deep sleep
​ ​
would also be a plus....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

cyberman54, 2018-03-22:
This is an open source project, you are welcome to contribute with pull requests.

I'm not able to build a framework from it, due to missing time and knowledge.

--

Does the code now compile and run with TTGOv1?

cyberman54, 2018-03-22:
RTC via TTN: it's not as easy to get exact time sync via LoRaWAN network. As far as i know github user FlorianLudwig has a solution on this, you may ask him.

https://github.com/FlorianLudwig

cyberman54, 2018-03-22:
Deep sleep would be an interesting option, but not for use case wifi scanner. Wifi scanning draws lots of power while running.
The Heltec and TTGO boards are not suitable for deep sleep, because the peripherals cannot be shutdown on these boards, that means they draw power alle the time.
Not sure yet what deep sleep capabilities the Pycom LoPy boards have.

trlafleur, 2018-03-22:
I let you code run all night, what interesting is that I see more join
request that data sends??  very odd...

- 08:06:06
dev addr:26022554app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 08:04:25
dev addr:26022629app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 08:04:18
dev addr:26022805app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 07:56:390
- 07:56:3901
retry
payload:00000000
- 07:56:34
dev addr:26022B04app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 07:55:280
- 07:45:4401
payload:00000000
- 07:55:28
dev addr:26022BADapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 07:22:570
- 07:22:57101
payload:00020000
- 07:18:560
- 07:18:5591
payload:00010000
- 07:14:540
- 07:14:5481
payload:00010000
- 07:10:530
- 07:10:5271
payload:00010000
- 06:44:00
dev addr:2602218Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:32:21
dev addr:26022E19app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:31:58
dev addr:260221ACapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:31:58
dev addr:2602217Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:22:31
dev addr:260229D8app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:21:46
dev addr:26022A5Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:20:250
- 06:18:3311
payload:00020000
- 06:20:250
- 06:15:4701
payload:00000000
- 06:20:25
dev addr:26022A43app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:20:250
- 06:13:1611
payload:00010000
- 06:10:340
- 06:09:4801
retry
payload:00000000
- 06:10:34
dev addr:26022D0Eapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:10:340
- 06:05:1401
payload:00000000
- 06:10:34
dev addr:2602222Eapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 06:04:51
dev addr:26022AC3app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:48:02
dev addr:2602295Eapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:46:55
dev addr:26022B60app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:05:22
dev addr:2602242Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:04:54
dev addr:26022093app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:04:54
dev addr:26022C9Dapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:04:54
dev addr:26022081app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 05:04:540
- 04:58:3221
payload:00020000
- 05:04:540
- 04:54:3011
payload:00010000
- 04:50:560
- 04:50:5701
payload:00000000
- 04:50:51
dev addr:26022440app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 04:50:28
dev addr:26022507app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:47:32
dev addr:26022E3Fapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:47:32
dev addr:260222F3app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:47:32
dev addr:26022D0Eapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:17:00
dev addr:26022444app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:17:00
dev addr:260225A6app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 03:13:5931
payload:00000000
- 03:17:000
- 03:09:5821
payload:00000000
- 03:05:560
- 03:05:5611
payload:00010000
- 03:03:040
- 03:03:0401
payload:00000000
- 03:02:59
dev addr:26022CF2app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:28:13
dev addr:26022A5Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:25:48
dev addr:2602216Dapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:25:480
- 02:24:1111
payload:00010000
- 02:25:480
- 02:22:0801
payload:00000000
- 02:25:48
dev addr:26022CB9app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:10:06
dev addr:26022270app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:09:09
dev addr:26022251app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:07:26
dev addr:26022219app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 02:07:26
dev addr:26022F2Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:50:47
dev addr:260226E9app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:45:50
dev addr:26022662app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:45:50
dev addr:2602275Capp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:45:50
dev addr:26022619app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:45:500
- 01:41:3011
retry
payload:00010000
- 01:38:17
dev addr:26022491app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:36:46
dev addr:26022323app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:36:46
dev addr:2602234Capp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:36:46
dev addr:260228E7app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:36:46
dev addr:26022DC3app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:36:46
dev addr:26022D70app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:21:19
dev addr:260225F6app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:21:19
dev addr:26022B47app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:21:19
dev addr:26022885app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:15:20
dev addr:2602273Bapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:15:20
dev addr:26022BF4app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:15:20
dev addr:26022220app eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:15:200
- 01:09:0511
payload:00010000
- 01:15:200
- 01:06:5701
retry
payload:00000000
- 01:15:20
dev addr:2602258Fapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:04:38
dev addr:260224EBapp eui:70B3D57EF0001E40dev eui:30AEA44CAAE0FEFF
- 01:04:03

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

cyberman54, 2018-03-22:
That looks like the device makes resets and then tries again to join the network. After a reset payload "0000" is sent.

Multiple joins may be caused by to weak LoRa radio coverage, or can be caused by bad RF-path on the ESP32 module. Since you seem to use a TTGOv1 you may want to take a look in TheThingsNetwork forum, multiple issues with bad RF performance were reported for TTGOv1. E.g. sometimes the pigtail cables were broken, or wrong antennas (433 MHz) were delivered.

Multiple resets can be caused intentionally by code, the code has a max LoRa retry threshold value which can be configured in main.h

Resets may also be caused by insufficient power supply.

The used LMiC code is robust and tested, it will sure not cause this behaviour.

trlafleur, 2018-03-22:
I have been using LMiC code for over a year and have over 220 nodes
deployed...

I agree with all of your points, but the GW ia 10 ft away, same TTGO
module, antenna, power supply has been running for over two week solid with
my base code....

So just wanted to point this out to you...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

cyberman54, 2018-03-22:
Wifi & BLE may need much more power than your base code on the TTGO.

My TTGOv2 board crashes when run on battery at the point in code, when Wifi & BLE are activated. The ESP32 throws a brownout reset.

Check the logs on serial console in verbose mode at the point, when the module resets. If it throws a brownout error, the problem is likely a power supply problem.

I can't guarantee that LMIC in RTOS task is running stable. I found reports on the net that say it does not. But i did long time tests over serveral weeks with 5 Heltec boards and this code, and they did not crash.

trlafleur, 2018-03-22:
Sorry for doing this in email.... I’m not on my computer...

This maybe an issue...

In main.c you do a:  #ifdef DEVEUI To see if DEVEUI is defined in you
config file...
A quick test of this by defining DEVEUI, shows it’s not working by sending
the generated DEVEUI.

No time to investigate this today....

void os_getDevEui (u1_t* buf) {
#ifdef DEVEUI
memcpy(buf, DEVEUI, 8);
RevBytes(buf, 8); // TTN requires it in LSB First order, so we swap bytes
#else
gen_lora_deveui(buf);
#endif
}

cyberman54, 2018-03-22:
please find a solution for this a make a pull request. I'm beginner im C / C++.

cyberman54, 2018-03-23:
fixed with 1.2.33

> In main.c you do a:  #ifdef DEVEUI To see if DEVEUI is defined in you
> config file...
> A quick test of this by defining DEVEUI, shows it’s not working by sending
> the generated DEVEUI.