cyberman54/ESP32-Paxcounter #7: Fresh platform.io setup : does not compile

hallard, 2018-03-22:
I tried to give a try with platformio, thinking it will be simple (installed from scratch), but I'm facing this problem.
First I needed to install the Bluetooth library by hand with
`platformio lib install 1841`

now lib is found but no luck, I'm facing this issue (esp32 idf version too recent)
https://community.platformio.org/t/how-to-set-up-esp32-sdk-toolchain/3464/2
Any help ?

cyberman54, 2018-03-23:
I can't give platformIO support here. If the problem is related to the bluetooth lib, you may try to comment out the Bluetooth switch in main.h, then no BT libs are needed.

cyberman54, 2018-03-23:
added in platformio.ini:
`lib_deps ESP32 BLE Arduino@>=0.4.9`

cyberman54, 2018-03-23:
Did you try a manual platform update?

> PlatformIO will install the latest Arduino Core for ESP32 from https://github.com/espressif/arduino-esp32. The Git should be installed in a system. To update Arduino Core to the latest revision, please open PlatformIO IDE and navigate to PIO Home > Platforms > Updates.

http://docs.platformio.org/en/latest/platforms/espressif32.html

closed 2018-03-23 by cyberman54

hallard, 2018-03-23:
@cyberman54 
Thanks for adding library in deps (I had done manually) but it's better now
I've done `navigate to PIO Home > Platforms > Updates`
Then it installed the latest esp32 and that worked, I still don't understand why yesterday it installed an old version of idf/tool chain
thanks, now that works ;-)

hallard, 2018-03-25:
I'm using ttgoV1 and heltec32, both compile and upload ;-)

hallard, 2018-03-25:
But I'll give a try with a fork to support custom shield I made for other ESP32 boards
https://github.com/hallard/LoLin32-Lora