cyberman54/ESP32-Paxcounter #34: Building errors in BLEDescriptor.cpp and in BLEClient.cpp

henkdebruijn, 2018-04-12:
I get the following errors when trying to build on Platformio:

```
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEDescriptor.cpp: In member function 'void BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEDescriptor.cpp:158:53: error: 'struct esp_ble_gatts_cb_param_t::gatts_add_char_descr_evt_param' has no member named 'descr_uuid'
m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&
^
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':
.piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:115:2: error: too many arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, bool)'
);
^
In file included from .piolibdeps/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:13:0:
/Users/henkdebruijn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bluedroid/esp_gattc_api.h:293:11: note: declared here
esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, bool is_direct);
```
I use the following configuration:

```
[env:lopy]
platform = espressif32
board = esp32dev
framework = arduino
monitor_baud = 115200
upload_speed = 921600
lib_deps = 
    U8g2@>2.21.7
    ESP32 BLE Arduino@>=0.4.9
    SmartLeds
build_flags = 
;set log level, we need build_flag for this, otherwise we can't use ESP_LOGx in arduino framework
    -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO
;   -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
;   -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_NONE
;needed for ESP32 BLE Ardunio v0.4.9
    -fexceptions
    -std=c++11
;override lora settings from LMiC library in lmic/config.h and use main.h instead
    -Dlopy
    -D_lmic_config_h_
    -include "src/main.h"
    -include "src/hal/lopy.h"
;FreeRTOS single core operation, switches off core 1 (see arduino-esp32/cores/esp32/main.cpp)
;   -DCONFIG_FREERTOS_UNICORE
```

cyberman54, 2018-04-12:
These are errors from inside the BLE stack, not from the Paxcounter code. Here it compiles without errors, have no clue how to reproduce this.

```
Compiling .pioenvs\lopy\lib5f6\ESP32 BLE Arduino_ID1841\BLEClient.cpp.o
Compiling .pioenvs\lopy\lib5f6\ESP32 BLE Arduino_ID1841\BLEDescriptor.cpp.o

```
Which version of Espressif32 library do you have installed? It should be 0.12.0

You could try to raise an issue here:

https://github.com/nkolban/ESP32_BLE_Arduino

or in Platformio community.

henkdebruijn, 2018-04-12:
[Solved] I used an older version of the ESP32 BLE library: `ESP32 BLE Arduino@0.4.9`

cyberman54, 2018-04-12:
hmm, i thought having covered this by using the libs_dep line in platformio.ini with version requirement. Didn't this force PlatformIO to update your BLE lib?

cyberman54, 2018-04-12:
Looks like you ran in this error:

https://github.com/nkolban/esp32-snippets/issues/468

closed 2018-04-12 by cyberman54

henkdebruijn, 2018-04-13:
```
;platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
```

Working, building is successful.