cyberman54/ESP32-Paxcounter #38: 433Mhz Heltec LoRA

Anonomousxxx, 2018-04-16:
Hi, we have 433Mhz devices, what can we do?

cyberman54, 2018-04-16:
You need a LoRaWAN Gateway, that it capable for 433 MHz and configured with a 433 MHh suitable channel plan. And you need a suitable LoRaWAN network, which supports this channel plan.

If you have both, please give me the channel plan, than i will implemement it.

Anonomousxxx, 2018-04-17:
Can one of the devices as the gateaway?

cyberman54, 2018-04-17:
Only as single channel gateway, what makes things more complicated and is not compliant to LoRaWAN specs. I wouldn't recommend this.

Anonomousxxx, 2018-04-17:
Well, I guess we will give at a shot for now, because we can't get a gateway quickly.
We found this https://github.com/kersing/ESP-1ch-Gateway-v5.0 for a gateway.
Also we found http://wiki.dragino.com/index.php?title=Connect_to_TTN#How_To_add_other_frequency_band_such_as_EU433 - but it says it would work only for ABP mode
Anyway, can you please tell what should be changed in the code to support 433mhz, so we could change channels by ourselves later for example?

cyberman54, 2018-04-18:
You need to change this in the LMiC library, not in the paxcounter code. Look here:
https://github.com/matthijskooijman/arduino-lmic/issues/98

closed 2018-04-18 by cyberman54