cyberman54/ESP32-Paxcounter #41: v1.3.0 crashes at a certain amount of count, around 800-1000

cyberman54, 2018-04-16:
Tested on TTGOv2, devices resets after reaching around 800-1000 counts, for yet an unknown reason.
Assumption: RAM oder stack memory leakage - to be figured out.

hallard, 2018-04-23:
Could be interesting to test with a LoPy4 which has 4Mb of PSRAM ;-)

cyberman54, 2018-04-23:
I'm currently exploring the memory situation. It seems that lack of heap memory causes the reset. The root cause seems to be the enormous memory consumption of std:set, 1 hash entry of uint16_t size cost 36 bytes. There are around 56KB free heap with actual version 1.3.2 on a Heltec. That means that heap is full after ~1550 hashed MACs.

hallard, 2018-04-23:
whooo, I know why I love C and not using cpp, I was not aware of std:set until I saw your code (but like it). We just need to know it's not memory efficient. I used to use C linked list for this, it takes a pointer (4 bytes here) on next element entry and 2 bytes for hash. 
Since alignment is on 4 bytes on ESP32 we could set the hash to 4 bytes and with linked list each element should take 8 bytes instead of 36.
we could even optimize, declaring an array of elements, IE a pointer point to 16 element like this cost of a pointer is not the same than the data stored (hash)

I already wrote something with linked list here https://github.com/hallard/remora_soft/blob/master/linked_list.cpp

cyberman54, 2018-04-24:
Interesting hint on the LoPy4, i did not recognize that this board comes with +4MB RAM for only €3 more. I already have own and will try to figure out how the extra memory can be used.

Parallel the software needs to be optimized for storing the hashes in memory for less cost.

cyberman54, 2018-04-24:
This [sparsepp](https://github.com/greg7mdp/sparsepp) could be a solution with good fit.

cyberman54, 2018-04-28:
memory threshold added to prevent reset by heap overflow

closed 2018-04-28 by cyberman54