cyberman54/ESP32-Paxcounter #54: OTAA JOIN not working

webfrank, 2018-04-30:
Hi,
   I've compiled and uploaded latest code on an Heltec Lora 32 but although a JOIN packet is sent to Gateway and the server replies on the node there is no EV_JOINED and on display I have:

Join wait
PACKET QUEUED

But from code I would expect JOINING

cyberman54, 2018-04-30:
Check your Network credentials and gateway configuration. The code was tested and runs on Heltec LoRa 32 with 868 MHz.

webfrank, 2018-05-01:
Hi, credentials are ok, there should be something on the receiving of lora data which is not working but the board I'm using was previously used as a 1ch gateway without problem. I'll investigate further.

cyberman54, 2018-05-01:
what kind and brand/type of gateway are you using?
What network are you using?

webfrank, 2018-05-01:
I have a Kerlink femtocell, with other devices (watteco temp sensor) with otaa works correctly. I can see from console that join accept message is sent and also with an sdr monitor I can see the transmission. It looks like there is a timing problem on Rx windows. 

Inviato da iPhone

cyberman54, 2018-05-01:
could also be a timing problem with the gateway. What LoRaWAN network are you using, TTN? 
TTN delays downlinks to a maximum to give slower gateways a greater chance. But this can cause problems when WAN connection of gateway has latency, e.g. because uplink is mobile phone network connection.

cyberman54, 2018-05-01:
try enabling LMiC debug messages (level 1 or level 2) in paxcounter.conf by uncommenting this line:

//#define LMIC_DEBUG_LEVEL 1

Then check LMiC timestamps and compare timing with RX settings of your network provider.

cyberman54, 2018-05-11:
Did you find a solution, can i close this issue?

closed 2018-05-11 by cyberman54