cyberman54/ESP32-Paxcounter #56: We want to cooperate with you

LilyGO, 2018-05-10:
Hello, Mr. cyberman54. We are a manufacturer of TTGO T3 VI.0 and V2.0. Given your previous project, we want to further cooperate with you. Correspondingly, we will pay you accordingly. If you are interested, please email laiqiuxia@lilygo.cc to discuss details with us.

aapris, 2018-05-10:
Hallo, also from Finland, Mr. cyberman54! We have a bunch of TTGO T3 V2.0 LoRa modules and we are going to use them as a sensor platform at Cable Factory (which is a huge cultural centre in Helsinki https://www.kaapelitehdas.fi/en/info). We measure environmental conditions (co2, humidity, temperature, noise etc) and use LoRaWAN to deliver the data to the servers and further use.

Because all modules are not in use yet, I installed Paxcounter in a few and got interesting data out from them already. It would be very interesting to use as visitor counters inside Cable factory, since nobody really have an estimate how much people there visits daily, weekly or annually.

We could help by testing the software! We just orderer 50 pieces more these modules (2 weeks ago! Where they linger @LilyGO ? :) and I can set up 10 modules running quickly at Cable factory. 

Just tell me how we can help?

LilyGO, 2018-05-11:
Hello, Mr. cyberman54. 
Thanks for the information and the support to our products. Let's talk more details by email. We will contact you via laiqiuxia@lilygo.cc

LilyGO, 2018-05-11:
@aapris Hello, Mr. cyberman54. 
Thanks for the information and the support to our products. Let's talk more details by email. We will contact you via laiqiuxia@lilygo.cc

aapris, 2018-05-11:
@LilyGO Our order just arrived today and next week I'll set up a set of Paxcounters and do some fields testing.

cyberman54, 2018-05-11:
@aapris Nice to hear of your big scale installation in cable factory! 

If you want to focus on tests, i suggest you use the [TTNmon](https://ttnmon.meis.space/console/#) tool to check if your nodes run stable and the paxcounter software does not cause unsolicited resets, e.g. due to memory lacks or watchdog triggers.

If you want to contribute to this project, please focus on creating a nice front end by integrating paxcounter in an IoT platform, e.g. Cayenne. We still miss a frontend completely here, since i focused on development of the embedded software so far.

aapris, 2018-05-11:
We use [Everynet's](https://everynet.com/) LoRaWAN and I'm not sure is it compatible with TTNmon tool. Currently we put the data into InfluxDB and use Grafana as a frontend. No further analysis is done yet, just graphs. Cable factory management/staff is anyway very interested in this, so I'll do more experiments during next weeks/months.

closed 2018-06-03 by cyberman54

gddeen, 2022-10-02:
Dear Sir,
I bought around 20 LilyGo boards (T-Beam V1.1, Lora  T3_V1.6, etc) for a project. 

I've since found that I need a LoRaWAN gateway before I can start playing with them. China is a LONG way away for getting one. It will take 3 weeks, plus they are on vacation this week.

Is there a github repository for something that could run on a LiLyGo board, or a Raspberry Pi4 that could receive these packets (as if a gateway, or how to convert Paxcounter to send with a set SF and unique ID) so I could focus on the other parts of the project? IoT (client) --> dummy IoT device -> WiFi -> MQTT or App.

I'm not all that interested in the aspects of the WAN (encryption, packet loss, Error correction, optimized spread, ADR, firmware updating).
I was looking at the code PAXCounter and where I could clip the WAN out of the project. I figured someone already did this when they wrote it! Thanks for any help.  And if no help, don't sweat it. I'll figure something out.

 gddeen@gmail.com

cyberman54, 2022-10-02:
You need hardware with a Lora concentrator chip, without this you won't be able to run a full 8ch lorawan compliant gateway. Using node hardware you can build a single channel lora gateway, but this is not lorawan compliant. Thus, you need to get a lorawan gateway.

cyberman54, 2022-10-02:
If you don't want to use lorawan, you easily can #define Has_lora 0 in paxcounter.conf.

gddeen, 2022-10-11:
Thanks.
There's a gateway 24 miles from me. Maybe I can find a coffee shop near
there for a month. I KNOW the guy will get excited when he starts seeing
some traffic.
I try to code around spurts of energy.

It is amazing this is 1.3 MB compiled. We were putting people on the 🌓
moon with less than a 16th of that! - Thanks