cyberman54/ESP32-Paxcounter #60: Persist LoRaWAN keys after OTAA join

cyberman54, 2018-05-18:
[HowTo:](https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973/97)

> After an OTAA Join has finished, your node has the very same details you’d have for an ABP device. So: store the DevAddr and the secret session keys NwkSKey and AppSKey, and keep track of both frame counters. After deep sleep, configure your node as an ABP node using LMIC_setSession(...), and set the frame counters to their last stored values, using LMIC.seqnoUp = ... and LMIC.seqnoDn = ....
> 
> For LMiC, these session details have different names (which originate from the early days where LoRaWAN was still called LoRaMAC). The documentation shows:
> 
> 2.5.4 void LMIC_setSession (u4_t netid, devaddr_t devaddr, u1_t* nwkKey, u1_t* artKey)
> 
> Set static session parameters. Instead of dynamically establishing a session by joining the network, precomputed session parameters can be provided. To resume a session with precomputed parameters, the frame sequence counters (LMIC.seqnoUp and LMIC.seqnoDn) must be restored to their latest values.
> 
> And so does the code:
> 
> void LMIC_setSession (u4_t netid, devaddr_t devaddr, xref2u1_t nwkKey, xref2u1_t artKey);
> Indeed, the global LMIC variable provides access to properties such as LMIC.nwkKey for NwkSKey, LMIC.artKey for AppSkey and LMIC.devaddr for DevAddr, which you should be able to retrieve and persist in case EV_JOINED.
> 
> Beware that you might also need to store and set the frequency and RX2 details that you get in the Join Accept, or after subsequent ADR commands…
> 
> (I have not used this.)
> 
>

stkdiretto, 2018-06-03:
I am currently looking into this since I could need this in another project also. Do you suggest storing the data in the `configdata_t` struct? I'd check for set parameters on reboot and using them if available.

cyberman54, 2018-06-03:
Could be a way. Another, maybe more straight forward way, would be to create/open a separare, second partition in NVRAM and store it there.

stkdiretto, 2018-06-04:
I have to dig into the storing mechanism a bit more. Your suggestion sounds more feasible – from what I've interpreted so far, saving the data would always mean storing _all_ the variables. Since `seqnoDn` and `seqnoUp` need to be saved on every package, that might wear out the nvram rather quickly.

FlorianLudwig, 2018-06-12:
To reduce write wearing only persist counters every `n` packages and when rejoining just add `n` to the counter.  This way the join works and the network still accepts the packages and regard the counter difference as package loss.

`n` should depend on send interval.

cyberman54, 2019-05-02:
This lib could / should be useful here:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.h

cyberman54, 2019-10-13:
Example how to save keys in RTC RAM of ESP32 while deep sleep:

https://github.com/manuelbl/ttn-esp32/issues/15

cyberman54, 2019-11-19:
RTC RAM save keys is now implemented.

closed 2019-11-19 by cyberman54