cyberman54/ESP32-Paxcounter #62: Make sure there is no packet loss while Wifi scanning

cyberman54, 2018-05-18:
On esp32.com forum i found some posts where packet loss is reported, while capturing packets in promiscuous mode. We should validate that our code doesn't have this problem, or, if there is a packet loss, we should try to calculate the quote. Setup could be a surrounding with defined Wifi traffic in the air, then capturing with paxcounter and parallel on a PC with wireshark, then compare results.

[Post packet loss 1](https://www.esp32.com/viewtopic.php?f=13&t=4145)
[Post packet loss 2](https://www.esp32.com/viewtopic.php?f=13&t=2517&p=11845&hilit=promiscuous#p11845)

cyberman54, 2019-10-26:
Rework packet parser `wifi_sniffer_packet_handler()` in wifiscan.cpp using logics from this [template](wifi_sniffer_packet_handler) to increase counting precision

Nethranv, 2019-11-27:
Please reshare the template , its not opening.

cyberman54, 2019-11-29:
@Nethranv > Please reshare the template , its not opening.

Link updated.

cyberman54, 2020-11-10:
Done with v2.0.4.

closed 2020-11-10 by cyberman54