cyberman54/ESP32-Paxcounter #65: on the ttgo v2.1 i always get this error : program size > maximum allowed

baxtery, 2018-05-30:
Error: The program size (1340950 bytes) is greater than maximum allowed (1310720 bytes)

cyberman54, 2018-05-30:
In PlatformIO, make sure you have Espressif32 0.12.0 release platform installed. NOT the staging version.

cyberman54, 2018-05-30:
Right now i recognized that today, just 4 hours ago, a new version 1.0.0 of platformio Espressif32 platform was released:
https://github.com/platformio/platform-espressif32

I had no time yet to compile and tes the paxcounter code against this release. If this produces an .elf file which is ~300kb bigger than with 0.12.0 i currently don't see any other way to use the 0.12.0 version.

I will modify paxcounter.ini later, to force this, if required.

cyberman54, 2018-05-30:
Meanwhile i tested compilation with the new Espressif32 v1.0.0 library.
The code compiles with success, but with some warnings (which are probably to be checked).
But it generates a binary sized ~1,4MB which is too large to fit in the 1,25 MB flash memory of the ESP32.

So, currently the paxcounter code is unfortunately not running with Espressif32 1.0.0 and i currently have no solution to solve this issue. I raised a question on community.platformio.org how to solve this.

Meanwhile it is possible to still use the older 0.12.0 library. I modified paxcounter.ini which now enforces PlatformIO to load the 0.12.0 version.

baxtery, 2018-05-31:
Thanks @cyberman54 I am amazed by the fastness :) will use the 0.12

closed 2018-05-31 by baxtery