cyberman54/ESP32-Paxcounter #68: TTN Console Payload Format Decoder

robbi5, 2018-06-04:
Thanks for the very nice project. For small tests and checking the pax count in the TTN console without converting the payload every time by hand, I've used these two functions for the Custom Payload Format settings:

#### decoder
```js
function Decoder(bytes, port) {
  // Decode an uplink message from a buffer
  // (array) of bytes to an object of fields.
  var decoded = {};

  if (port === 1) {
    decoded.wifi = (bytes[0] << 8) | bytes[1];
    decoded.ble = (bytes[2] << 8) | bytes[3];
  }

  return decoded;
}
```

#### converter
```js
function Converter(decoded, port) {
  // Merge, split or otherwise
  // mutate decoded fields.
  var converted = decoded;

  if (port === 1) {
    converted.pax = converted.ble + converted.wifi;
  }

  return converted;
}
```

Feel free to improve/put them in the readme/whatever :)

cyberman54, 2018-06-04:
Works like a charme, thank you.
I will integrate this in the description some where.

cyberman54, 2018-06-04:
Code integrated in readme.md

closed 2018-06-04 by cyberman54