cyberman54/ESP32-Paxcounter #76: time based pax counter reset

FlorianLudwig, 2018-06-10:
Currently the counter is reset after a successful (but not acknowledged) transmission of the counter values. This means that the time frame in which for a counting cycle is not constant making it harder to Interpreter the results. Therefor I like to suggest decoupling the sending of counter values and resetting the counter.

FlorianLudwig, 2018-06-10:
I think it's best to use the same timer but reset the counter unconditionally wehter the packet was sent or not.

Having sending/sampling the data and resetting the counter not aligned doesn't have any use case I can think of.

cyberman54, 2018-06-10:
This is the status quo. Currently the counter is always reset when the data was enqueued to send, except in consecutive mode.

// clear counter if not in cumulative counter mode

if (cfg.countermode != 1) {

reset_counters(); // clear macs container and reset all counters

reset_salt(); // get new salt for salting hashes

ESP_LOGI(TAG, "Counter cleared (countermode = %d)", cfg.countermode);

}

FlorianLudwig, 2018-06-10:
that part only runs when
```
if (LMIC.opmode & OP_TXRXPEND) {
```
is not the case. 

I would like to move it outside of that check.

cyberman54, 2018-06-10:
Oh, this is a bug, not intentionally. Please change it.

commit 4fce10b0 (2018-06-10): always reset mac counters even if lora package could not been sent

fixes #76
closed 2018-06-10 by FlorianLudwig