cyberman54/ESP32-Paxcounter #101: W (565) wifi: ap start fail

qvs5010, 2018-06-26:
Hi 

I just got the TTGO T-Beam, and did the changes in the config files, but when the board boots up I keep getting `W (565) wifi: ap start fail`

Is there a place I must specify the wifi?
Thanks

```
[I][main.cpp:622] setup(): Starting Lora task on core 1
[I][main.cpp:627] setup(): Starting Wifi task on core 0
I (194) wifi: wifi firmware version: 2fc15c0
I (197) wifi: config NVS flash: disabled
I (200) wifi: config nano formating: disabled
I (206) wifi: Init dynamic tx buffer num: 32
I (209) wifi: Init data frame dynamic rx buffer num: 64
I (214) wifi: Init management frame dynamic rx buffer num: 64
I (219) wifi: wifi driver task: 3ffdac68, prio:23, stack:4096
I (225) wifi: Init static rx buffer num: 10
I (228) wifi: Init dynamic rx buffer num: 0
I (560) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=1

W (565) wifi: ap start fail

Brownout detector was triggered
```

cyberman54, 2018-06-27:
wifi ap start fail is normal, just ignore. But the brownout detection causes a reset. Check your power supply or enable brownout protection in ttgo-beam.h; see ttgov2.h

cyberman54, 2018-06-27:
this line must be added to ttgo-beam.h:

#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature

But on my TTGO T-Beam it works without.

qvs5010, 2018-06-27:
Hi, it was Cables, just bad usb cables, agggg, I had 4 brand new cables and they failed, got a new james bond usb cable today and now all is working, thanks!!!

cyberman54, 2018-06-27:
so could please close your issue?

closed 2018-06-27 by cyberman54

qvs5010, 2018-06-27:
Apologies, didn't know I had the power close. ;)