cyberman54/ESP32-Paxcounter #103: src/macsniff.cpp:15:75

sbamueller, 2018-07-03:
[image]

Now I get this issue

cyberman54, 2018-07-03:
On MS VSC with PlatformIO core this compiles without error.
I can't solve this, we need some C/C++ expert here.

sbamueller, 2018-07-03:
The information system student next to me says your are doing fancy shit which could be replaced by simpler code.

Alternative does it work on Microsoft Windows?

cyberman54, 2018-07-03:
If he or you can do better, make pull request.

Meanwhile i fixed Espressif Platform to 1.0.2, perhaps you used a newer one.
Try again to compile after syncing master.

cyberman54, 2018-07-03:
Workaround by fixing Espressif32 core to v1.0.2 works.
Problem occurs after update Espressif32 core to v1.1.0.

cyberman54, 2018-07-03:
solved in v1.8.83 (currently available only on branch development)

closed 2018-07-03 by cyberman54