cyberman54/ESP32-Paxcounter #109: paxcounter SENSOR: implement I2C and/or SPI interface

cyberman54, 2018-07-16:
Make a paxcounter SENSOR variant (besides paxcounter LORA): Make it usable for integration in devices and sensor appliances.

- implement I2C and/or SPI interface to connect paxcounter sensor to a wired host
- I2C uses same set/get commands as in rcommand.cpp
- Make I2C devices address configurable by hardware: reading hardwired binary settings from 3 GPIOs
- SPI using simple data structure: Data out = port -> payload / data; Data In = port -> rcommand
- Using SPI for firmware update OTA

cyberman54, 2018-10-04:
@der-ambi spi.cpp  / spi.h are now ready-to-use prepared in development branch ;-)

der-ambi, 2018-11-02:
Work is nearing completion, progress can be seen [in my fork](https://github.com/der-ambi/ESP32-Paxcounter/tree/spi_on_development_3110)

cyberman54, 2018-11-02:
@der-ambi looks good. 
I had not in mind that we need to behave as SPI slave, not master.

gregorwolf, 2018-11-22:
Is there any documentation to what GPIO Pins the SPI / I2C needs to be connected?

cyberman54, 2018-11-22:
No, since this depends on your hardware and you must configure this on yourself.
Pins are configured in the hardware's hal file, e.g. in `hal/generic.h`

```
// pin definitions for SPI slave interface
#define SPI_MOSI GPIO_NUM_23
#define SPI_MISO GPIO_NUM_19
#define SPI_SCLK GPIO_NUM_18
#define SPI_CS   GPIO_NUM_5
```