cyberman54/ESP32-Paxcounter #113: Low power support

younesmaia, 2018-07-21:
- Enable a configurable low-power mode, in which the device only wakes up after a set interval in order to scan for MACs and BSSIDs (that would be great for indoor location), get a location fix from the GPS, transmit everything and then sleep again.

- Add support for a flag set by the gateway that prevents the device from sleeping for a set period (for on-demand scanning).

proffalken, 2019-02-26:
Agree that this would be excellent - perhaps the work at https://github.com/JoepSchyns/Low_power_TTGO_T-beam would be useful here?

cyberman54, 2019-02-26:
@proffalken the linked example is not based on a LoRaWAN stack, it just transfers data by raw LoRa. So, besides the hardware programming for sleep mode, we will need additional code to leave/rejoin the LoRaWAN network when device goes to sleep / wakes up.

In general, low power programming must always be close to hardware, so it won't be possible to keep a hardware agnostic code for paxcounter. Not a must have, but will make code more complicated since hardware dependent switches/subroutines are needed.

I did not spend time on low power programming here, because wifi sniffing is an energy consumptive use case. But for battery powered devices it could be a reasonal solution to operate them during short intervals to catch a time stamped probe, then sleep again.

proffalken, 2019-02-26:
@cyberman54 ah, thanks, makes sense - I'm far more of an infrastructure guy than a hardware guy, so happy to learn from you on this one.

I'm thinking about adding a solar-panel to my T-Beam based Paxcounter to trickle charge the battery and see how long it lasts as well as upping the scan rate to once every couple of minutes ( we don't need more than that for my current use-case).

Thanks for the great work on the project though, after getting it working last night at home, I've had it working out and about this morning already! https://twitter.com/MBConsultingUK/status/1100322967135965184

cyberman54, 2019-02-26:
@proffalken when it comes to low power, ususally this is talking about devices in microampere range.
The TTGO boards do not come with the necessary design. It means not only to put the CPU (ESP32) in sleep mode, but also to power down all peripherals by software. This is e.g. not possible with most of TTGO boards for 
- the USB-UART chip (2102/2104)
- the pullup/pulldown resistors
- voltage divider to measure battery voltage
- the GPS and LORA chip

GPS and LORA chip can be reached by software and have own sleep modes which can activated by software. But with TTGO board this effort ist not worth to carry out, because the above peripherals still draw power.

10mA is still far to high for an autonomous device, powered by battery and/or solar. Even with a huge 18650 Li-Ion cell with 3000mAh you won't reach more than 10 days device life time.

Low power means life times, in my opinion, of at least 2 years.

The new HELTECv2 board and the TTGO-FOX board have slightly better design for low power. TTGO-FOX has a transistor which powers down peripherals by signal on GPIO.

But if you want real low power, you better go for other hardware, e.g. the Nexus by

proffalken, 2019-02-26:
@cyberman54 Understood, thanks. :)

cyberman54, 2019-09-23:
@proffalken @younesmaia We have some news here. Current paxcounter version now supports the AXP192/202 power management chip series, as found on TTGO T-BEAM V10 board. With this circuit we have better premises to implement a real low power mode.

cyberman54, 2020-12-09:
@younesmaia @proffalken The requested feature is currently implemented, see https://github.com/cyberman54/ESP32-Paxcounter/issues/627

closed 2020-12-09 by cyberman54

proffalken, 2020-12-09:
Amazing, thank you!