cyberman54/ESP32-Paxcounter #121: t-beam missing Time library

nalberto, 2018-07-23:
I am trying to compile for the TTGO T-Beam and it does not compile with the following error.
I have tried to include the Time library using the library manager but no improvement.
What am I doing wrong?
Help please!
`---------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
DEBUG: CURRENT(minimodule) EXTERNAL(minimodule, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyGPSPlus> 1.0.2
|-- <Time> 1.5
|-- <IBM LMIC framework> 1.5.0+arduino-2
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <U8g2> 2.23.12
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SmartLeds> 1.1.3
Compiling .pioenvs/ttgobeam/src/homecycle.cpp.o
Archiving .pioenvs/ttgobeam/lib492/libarduino-lmic-1.5.0-arduino-2-tweaked
Indexing .pioenvs/ttgobeam/lib492/libarduino-lmic-1.5.0-arduino-2-tweaked
Archiving .pioenvs/ttgobeam/lib0f6/libWire.a
Indexing .pioenvs/ttgobeam/lib0f6/libWire.a
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/U8x8lib.cpp.o
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_bitmap.c.o
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_box.c.o
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_buffer.c.o
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_circle.c.o
src/homecycle.cpp: In function 'void doHomework()':
src/homecycle.cpp:23:62: error: 'setTime' was not declared in this scope
gps.date.day(), gps.date.month(), gps.date.year());
^
In file included from /home/nicolas/.platformio/packages/framework-arduinoespressif32@1.5.3/cores/esp32/esp32-hal.h:49:0,
from /home/nicolas/.platformio/packages/framework-arduinoespressif32@1.5.3/cores/esp32/Arduino.h:35,
from src/globals.h:5,
from src/homecycle.cpp:5:
src/homecycle.cpp:24:57: error: 'hour' was not declared in this scope
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());

^
/home/nicolas/.platformio/packages/framework-arduinoespressif32@1.5.3/cores/esp32/esp32-hal-log.h:93:72: note: in definition of macro 'log_i'
#define log_i(format, ...) log_printf(ARDUHAL_LOG_FORMAT(I, format), ##__VA_ARGS__)
^
src/homecycle.cpp:24:5: note: in expansion of macro 'ESP_LOGI'
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());
^
src/homecycle.cpp:24:67: error: 'minute' was not declared in this scope
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());
^
/home/nicolas/.platformio/packages/framework-arduinoespressif32@1.5.3/cores/esp32/esp32-hal-log.h:93:72: note: in definition of macro 'log_i'
#define log_i(format, ...) log_printf(ARDUHAL_LOG_FORMAT(I, format), ##__VA_ARGS__)
^
src/homecycle.cpp:24:5: note: in expansion of macro 'ESP_LOGI'
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());
^
src/homecycle.cpp:24:77: error: 'second' was not declared in this scope
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());
^
/home/nicolas/.platformio/packages/framework-arduinoespressif32@1.5.3/cores/esp32/esp32-hal-log.h:93:72: note: in definition of macro 'log_i'
#define log_i(format, ...) log_printf(ARDUHAL_LOG_FORMAT(I, format), ##__VA_ARGS__)
^
src/homecycle.cpp:24:5: note: in expansion of macro 'ESP_LOGI'
ESP_LOGI(TAG, "Time synced to %02d:%02d:%02d", hour(), minute(), second());
^
*** [.pioenvs/ttgobeam/src/homecycle.cpp.o] Error 1
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_cleardisplay.c.o
Compiling .pioenvs/ttgobeam/libacc/U8g2_ID942-tweaked/clib/u8g2_d_memory.c.o
========================================================================================================== [ERROR] Took 2.80 seconds ============================================
==============================================================`

cyberman54, 2018-07-23:
I can't reproduce this, it compiles on my IDE without this issue.
Did you try a platformIO clean before build?

`platformio run --target clean` 

-> clean project (remove compiled files)

nalberto, 2018-07-23:
Ok I managed to load it. Trick is to switch off the board (onboard switch) before clicking on the upload button on platformio, wait 5 seconds and turn on the switch. That way the board is able to be flashed.

closed 2018-07-23 by cyberman54