cyberman54/ESP32-Paxcounter #122: Negative pax bug

SchneHa, 2018-07-24:
If count is set to 1, Cayenne payload can become negative. In OLED display count is shown correctly but Cayenne payload is wrong (negative).

cyberman54, 2018-07-24:
please inspect payload.cpp to check if this is a bug. Perhaps Cayenne does some calculations on their side for the analog input field, i don't know.

cyberman54, 2018-07-24:
can you figure out from the history data view in cayenne, at which number the counter turns to negative? Perhaps 254, 255 or 256?

SchneHa, 2018-07-24:
[image]

SchneHa, 2018-07-24:
I can't recognize a pattern

SchneHa, 2018-07-24:
I sent a message to the guys of Cayenne.

cyberman54, 2018-07-24:
This is the conversion in payload.cpp for Cayenne LPP encoding, using [simple arduino highbyte/lowbyte conversion](https://www.arduino.cc/reference/en/language/functions/bits-and-bytes/highbyte/):

```
buffer[cursor++] = LPP_ANALOG_INPUT; // workaround, type meter not found?
buffer[cursor++] = highByte(val1);
buffer[cursor++] = lowByte(val1);
```
This conversion does not regard signed values. But Cayenne is using a signed format:

```
Type | IPSO | LPP | Hex | Data Size | Data Resolution per bit
Analog Input | 3202 | 2 | 2 | 2 | 0.01 Signed

```
So i assume here is the bug. But what would be the right conversion for Cayenne, and what will be the highest possible value to show with it?

What we really need here would be simple 16bit "meter/counter" value field. But Cayenne does not provide this yet.

cyberman54, 2018-07-24:
Okay, this is obviously the bug.
For analog input, Cayenne awaits a signed 16-bit integer, and interprets this as value * 0,01.
Highest positive number which can be showed with a signed-bit16 integer is 32767.
32767 * 0,01 = 327,67 => 327 pax.
Values above 32767 will be interpreted as negative number:

No chance to get this fixed with Cayenne analog input.

So i will change data type to "illuminance sensor". Then Cayenne awaits an unsigned 16bit value in MSB format. This is what we need here. But will be displayed on Cayenne as "Lux", not "Pax".

I will change code in development branch in next hours.

cyberman54, 2018-07-24:
Thanks for testing. Please close issue if it now works.