cyberman54/ESP32-Paxcounter #124: Incorrect Sat count

SchneHa, 2018-07-25:
Showing sats on the display seems not really to work properly: If I got no GPS fix the value for sats is invers. That’s as you intended. If I am outside with the device and got a GPS fix, the number of sats is shown correctly (4 or 8 up to 12). If I go with this fix inside and have no sight to the sky, after a while the fix LED is still blinking but sat value shows 60 or 80, up to 90 satellites. That’s impossible.

cyberman54, 2018-10-06:
I still have no idea what's the root cause here.
Maybe this pull request would give a solution on it?
https://github.com/mikalhart/TinyGPSPlus/pull/38

cyberman54, 2018-12-28:
I attached an OLED display to a TTGO T-Team v07 with UBlox NEO 6m gps chip to reproduce this. 
I can confirm that after changing from outdoor to indoor the red led (driven by NEO 6m timepulse output) keeps blinking, but number of displayed Sats decreases to zero. In this situation the NEO 6m still delivers a valid time.

closed 2019-02-04 by cyberman54