cyberman54/ESP32-Paxcounter #127: Source for parts?

rdmueller, 2018-08-01:
as I am new to LoRa and arduino, I am looking for a good source of the right parts in germany.
any suggestion?

cyberman54, 2018-08-01:
you may try
https://www.exp-tech.de
https://eckstein-shop.de
https://www.ideetron.nl
https://pycom.io/webshop/

rdmueller, 2018-08-01:
thanx. which device would you recmmend?

rdmueller, 2018-08-01:
thanx. for this I need a display and usb board in addition? What kind of display can I use? are there ePaper display available for this device?

cyberman54, 2018-08-01:
Heltec Lora 32 Kit or TTGO v2.1 board is a good starting point if you need display. These boards have 
 OLED, LoRa and ESP32 on board for a low price. Sometimes you find sone on amazon.de or amazon.uk which are delivered from europe in 2 or 3 days, if not you can order them on aliexpress, look for LilyGo store there. Delivery from china with aliexepress standard shipping to Germany usually takes 10-14 days.

https://www.amazon.co.uk/MakerHawk-ESP8266-Development-Display-Support-As-Shown/dp/B076S8S6HL/ref=sr_1_fkmr0_2?ie=UTF8&qid=1533142359&sr=8-2-fkmr0&keywords=heltec+lora

I close this issuse hereafter, since this is not an issue. Please do not use the issue tracker system for general questions on LoRa / iot. Please use a forum instead, e.g.  https://www.thethingsnetwork.org/forum/

closed 2018-08-01 by cyberman54

rdmueller, 2018-08-01:
thanx for answering my beginner questions!
I very much apprechiate it!