cyberman54/ESP32-Paxcounter #133: Memory leak in v1.4.2

cyberman54, 2018-08-07:
Since v1.4.2 a memory leak occurs.
Not sure yet if this is a false positive.

der-ambi, 2018-08-08:
`git bisect` is very helpful for finding the problematic commit.

cyberman54, 2018-08-09:
I think i found the Problem, and can solve it now. Changing xQueues to pointers. Now testing.

cyberman54, 2018-08-09:
Pull Requests or Push is welcome here!

cyberman54, 2018-08-09:
Tried change to non global pointers. But each call still consumes 84 bytes 👎 

```
void processSendBuffer() {

  MessageBuffer_t *Queueptr, SendBuffer;
  Queueptr = &SendBuffer;

#ifdef HAS_LORA
  // Check if there is a pending TX/RX job running
  if ((LMIC.opmode & (OP_JOINING | OP_REJOIN | OP_TXDATA | OP_POLL)) != 0) {
    // LoRa Busy -> don't eat data from queue, since it cannot be sent
  } else {
    if (xQueueReceive(LoraSendQueue, &Queueptr, (TickType_t)0) == pdTRUE) {
      // SendBuffer gets struct MessageBuffer with next payload from queue
      LMIC_setTxData2(SendBuffer.MessagePort, SendBuffer.Message,
                      SendBuffer.MessageSize, (cfg.countermode & 0x02));
      ESP_LOGI(TAG, "%d bytes sent to LoRa", SendBuffer.MessageSize);
      sprintf(display_line7, "PACKET QUEUED");
    }
  }
#endif
```

pstorch, 2018-08-09:
Hi, I'm not an expert and don't have such a board to test my assumption, but an idea would be to free the memory allocated with `memcpy` in line 13. I would add `free(SendBuffer.Message);` at the end of the method `SendData()` in line 34.

cyberman54, 2018-08-10:
@pstorch Leak is in processSendBuffer(), not in SendData().

cyberman54, 2018-08-10:
@pstorch I tried to explicite `malloc `memory for the SendBuffer, and `free` it before exiting the function. But still each call of `processSendBuffer()` and `sendPayload()` consumes 84 bytes each.
```
// cyclic called function to eat data from RTos send queues and transmit it
void processSendBuffer() {

  MessageBuffer_t *SendBuffer;
  SendBuffer = (MessageBuffer_t *)malloc(sizeof(MessageBuffer_t));

#ifdef HAS_LORA
  // Check if there is a pending TX/RX job running
  if ((LMIC.opmode & (OP_JOINING | OP_REJOIN | OP_TXDATA | OP_POLL)) != 0) {
    // LoRa Busy -> don't eat data from queue, since it cannot be sent
  } else {
    if (xQueueReceive(LoraSendQueue, SendBuffer, (TickType_t)0) == pdTRUE) {
      // SendBuffer gets struct MessageBuffer with next payload from queue
      LMIC_setTxData2(SendBuffer->MessagePort, SendBuffer->Message,
                      SendBuffer->MessageSize, (cfg.countermode & 0x02));
      ESP_LOGI(TAG, "%d bytes sent to LoRa", SendBuffer->MessageSize);
      sprintf(display_line7, "PACKET QUEUED");
    }
  }
#endif

#ifdef HAS_SPI
  if (xQueueReceive(SPISendQueue, SendBuffer, (TickType_t)0) == pdTRUE) {
    ESP_LOGI(TAG, "%d bytes sent to SPI", SendBuffer->MessageSize);
  }
#endif

free(SendBuffer);

} // processSendBuffer
```

cyberman54, 2018-08-10:
I recently pushed new version in development branch:
- SendBuffer in SendData.cpp is now local, no more global. But this doesn't solve this mem leak issue.
- Added some debugging code for showing memory usage

pstorch, 2018-08-11:
https://www.freertos.org/xQueueSendToBack.html and https://www.freertos.org/a00118.html make copies of the payload.

Can you check whether a call to `flushQueues()` reduces the allocated memory? Then we know if the memory is held in the queues or in the Paxounter.

cyberman54, 2018-08-11:
@pstorch we don't want to flush the queues, if we would do this, usage of a queue would make no sense. The queue's task is to buffer data if it can't be transmitted on LoRa, after LoRa duty cycle is reached or a rejoin is ongoing.

According to [this post](https://www.freertos.org/FreeRTOS_Support_Forum_Archive/August_2009/freertos_Memory_Leak_3353184.html) "FreeRTOS only allocates memory when a task or queue is created, so a memory leak would seem doubtful."

cyberman54, 2018-08-11:
@pstorch @der-ambi Turns out: It seems it's just impact of the logging. 

ESP_LOGI(TAG, "%d bytes sent to LoRa", SendBuffer.MessageSize);

cyberman54, 2018-08-11:
solved with v1.4.23

closed 2018-08-11 by cyberman54