cyberman54/ESP32-Paxcounter #142: How to enable output?

provotector, 2018-09-06:
Hello. I'm using ESP32 dev board. When I build and flash all goes fine, but I have not output anywhere when the program is running on my board. How can I see the output? I'm trying to output to the Serial monitor but if I uncomment this line `#define LMIC_FAILURE_TO Serial` (at paxcounter.conf) the program is not compiling. I have this error: 

`lib\arduino-lmic-1.5.0-arduino-2-tweaked\src\hal\hal.cpp: In function 'void hal_printf_init()':
lib\arduino-lmic-1.5.0-arduino-2-tweaked\src\hal\hal.cpp:239:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope
fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
^
lib\arduino-lmic-1.5.0-arduino-2-tweaked\src\hal\hal.cpp:239:71: error: 'fdev_setup_stream' was not declared in this scope
fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);`

Is there an easy way to enable the output? Thanks!

der-ambi, 2018-09-06:
LMic belongs to LoRaWAN. If your board does not have LoRaWAN, you should not touch those settings.

When building current master with `env_default=generic` in `platformio.ini`, my personal ESP32 board happily logs on the serial console over USB.
Maybe you want to adjust `-DCORE_DEBUG_LEVEL=0` in `platformio.ini` to increase the debug level?

cyberman54, 2018-09-06:
Don't touch that setting. Serial console output is controlled by loglevel setting in paxcounter.conf, as descibed in previous post.

closed 2018-09-06 by cyberman54