cyberman54/ESP32-Paxcounter #148: I have "0" values at counters

provotector, 2018-09-12:
My real problem is that I have no counters. They show "0" any time at OLED screen. I tryed with 2 different boards (custom ESP32 board and Lolin32) and I have still the same problem. I tryed to establish a RSSI LIMITER only for screen debug purposes. When I modify this line (at configmanager.cpp) to for example -10

`cfg.rssilimit = -10;          // threshold for rssilimiter, negative value!`

The OLED screen is still showing "RLIM: off" after restarting the board. This is the normal behavior? Or I have some "general" problems that can affect the paxcounter is not counting and showing forever "0" values at counters? 

Thanks for the help and for sharing this really awesome sourcecode!!

disk91, 2018-09-12:
Sounds like I've the same issue with conf file.
For your specific problem of getting 0 I've got the same problem when the selected board was not the right version TTGO 2.1.x was in fact a TTGO 2.

cyberman54, 2018-09-13:
The device pulls it's runtime configuration from NVS. The values should be changed during runtime by using remote commands (see rcommand.cpp), the new setting is then automatically stored in NVS. Don't rebuild code everytime if you just want to change runtime parameters. Send an rcommand via your LORAWAN network instead to the device.

cyberman54, 2018-09-21:
Monitor mode controls beacon alerting, no affect on counters.
Unlikey RSSI has influence here, because by default there is no RSSI filtering.

I assume a side effect caused by issue #145 which occurs only with older ESP32 CPUs that have Revision 0 (not 1). You can check revision of your board in startup messages on console.

Today i could find a solution for issue #145, it is implemented in code in development branch.

Could you please flash this code and report here, if it solved your counter issue? Thanks.

cyberman54, 2018-09-30:
@provotector Please update software to current version and reopen this issue if problem persists.

closed 2018-09-30 by cyberman54