cyberman54/ESP32-Paxcounter #150: Update vendor_array.h to newer OUI list

spmrider, 2018-09-14:
An updated version of vendor_array.h has been created, it is available at [https://service.rules.org/owncloud/index.php/s/QQZN4eK3o3Zf3xu](url)

This version adds one vendor (Caterpillar, on special request) as well as updates the list with new prefixes for all other vendors

cyberman54, 2018-09-20:
Done.

spmrider, 2019-03-11:
Since last Update the vendor list changed again, I'll put up an updated version.

spmrider, 2019-03-11:
Done (committed to master)

spmrider, 2019-08-07:
New vendor_array.h has been committed to development branch

cyberman54, 2019-08-07:
Can we add toolchain to create vendor_array.h to the repository? (license, ... )

spmrider, 2019-08-07:
I just ported the toolchain from FreeBSD to Linux, needs some cleanup, will 
add it shortly to a tools folder

cyberman54, 2020-11-30:
Continued in https://github.com/cyberman54/ESP32-Paxcounter/issues/675

closed 2020-11-30 by cyberman54