cyberman54/ESP32-Paxcounter #156: Usage of Wire library.

Koepel, 2018-09-20:
In the file "ESP32-Paxcounter/src/lorawan.cpp", in the function `get_hard_deveui()`, the functions to read and write data with I2C are mixed together.

The first part has the sequence: Wire.begin - Wire.beginTransmission - Wire.write - Wire.endTransmission
That is good.

The second part has the sequence: Wire.beginTransmission - Wire.write - Wire.requestFrom - Wire.available - Wire.read - Wire.endTransmission
That is a mix between reading and writing. Writing data should always have `Wire.endTransmission()` at the end, and the `Wire.beginTransmission()` and `Wire.endTransmission()` may not be used together with `Wire.requestFrom()`.

See my [alternative explanation](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Explanation-of-the-functions-of-the-Wire-library).

cyberman54, 2018-09-20:
Thanks for the hints, i will look at this.
I assume I2C problems with the HAL from the display library, o8g2, too. Could you take at look at i2c driver in hal file of o8g2 lib, too?

Koepel, 2018-09-20:
Do you mean the u8g2 library itself? I think it is okay.
There is [only one](https://github.com/olikraus/u8g2/blob/feaf9c86aea07b4caf15162b0be2a0c9efc45e56/sys/arduino/u8g2_page_buffer/TrackControl/TrackControl.ino#L613) `Wire.requestFrom()` in u8g2, and that one is correct (link valid on sep 20, 2018). Everything else is writing data, and as far as I can tell, they are also correct.

cyberman54, 2018-09-20:
There is a long thread on some wired kind of interrupt problems possibly caused by i2c hal of u8g2. Indeed, display speed on paxcounter is very low since last updates.

https://www.google.de/url?sa=t&source=web&rct=j&url=https://github.com/espressif/arduino-esp32/issues/1588&ved=2ahUKEwi7ztmNmMndAhXGlCwKHZK7BS8QFjAAegQIAxAB&usg=AOvVaw38WAZsDhLH6rf8Jupf5mqd

Koepel, 2018-09-20:
Thanks, I will read it, but I don't think I can help them.
The ESP32 Wire-compatible library has changed, perhaps they still have few things to fix.

cyberman54, 2018-09-20:
i made changes in lorawan.cpp and gps.cpp according to your hints. Could you please re-check? You find the changes in the development branch. Thanks.

cyberman54, 2018-09-20:
oh, it should do still the same.

closed 2018-09-20 by cyberman54