cyberman54/ESP32-Paxcounter #171: Change LMIC to lmic esp32 port

cyberman54, 2018-09-30:
using [this lmic port f](https://github.com/ngraziano/lmicesp32)or esp32:

- persists LoRaWAN keys in RTC NVRAM
- uses ESP32 hardware AES acceleration

mutton74, 2018-10-08:
It would great to use this fork to allow for different band plans easily, having trouble using this code on AU915 at the moment
https://github.com/mcci-catena/arduino-lmic

hagenbuch, 2018-10-17:
Isnt the RTC RAM erased with every reset?

cyberman54, 2018-10-17:
Depends on kind of reset, RTC does not survive power cycle, we need NVS RAM for that.
Take a look here: https://www.esp32.com/viewtopic.php?t=4931

cyberman54, 2018-10-21:
Done with v1.6.5, code migrated to [LMIC 1.6 arduino port by MCCI](https://github.com/mcci-catena/arduino-lmic)

@mutton74 : you find this version currently in the development branch - could you please test it?

closed 2018-10-21 by cyberman54