cyberman54/ESP32-Paxcounter #174: add temp/humiditiy support

sbamueller, 2018-10-01:
There is one enhancement I would like to be able to add temp/humid/pressure to the sensor.
For example to monitor if the case is broken or see if the amount of people measured depends on the enviroment.

cyberman54, 2018-10-01:
There is no standard for sensors and interfacing them. You're welcome to develop a certain solution, e.g. with DHT11/21 or BM280 with SPI oder I2C interface, and make pull request here.

Ingetration can be done similary to GPS, take a closer look to gps.cpp.
Additional sensors should usw own RTOS task.

cyberman54, 2018-10-01:
> There is one enhancement I would like to be able to add temp/humid/pressure to the sensor.
> For example to monitor if the case is broken or see if the amount of people measured depends on the enviroment.

There is already a simple temperature monitoring implemented in paxcounter, using the onboard temperature sensor of ESP32 chip. The value can be requested by rcommand 0x81.
The temp sensor of esp32 ist not calibrated, so gives no absolute precision. But it works relative, e.g. it is possible to measure temperature curves.

hagenbuch, 2018-10-17:
I second sbamueller's suggestion - to me, it would be sufficient to point to a place where I could fit in I2C communication: I2C might already being used in the TTGO, maybe for non-volatile RAM? I've been browsing through your code (awesome project BTW! Thank you!) but haven't gotten an idea how to integrate something like that without risk of destruction. I'm rather new to embedded development, learned C (earned my first money with that) 33 years ago..

cyberman54, 2018-10-17:
look in gps.cpp, you find a simple i2c communication interface there.

sbamueller, 2018-11-18:
Ok let me specify: 
Transmitting data about the current temp/humidity and even some qir quality like VOC would be helpful in a transportation context. E.g. station service providers could see if temperature is below a certain value and it might freeze, thus send out someone to clear the station of ice or put gravel and salt on the street. 
Adding airquality might also be an issue in enclosed stations (e.g. underground) or when operating diesel or steam powerd trains. (see: [https://sbamueller.wordpress.com/2018/07/23/chanderli-macht-keinen-feinstaub-lifg-anfrage-bringt-klarheit/](https://sbamueller.wordpress.com/2018/07/23/chanderli-macht-keinen-feinstaub-lifg-anfrage-bringt-klarheit/)).
I suggest however we start with adding support for the BME680, which is already included in the Octopus Board build by Guido Burger and supported. Now having cheap Board (like TTGOBeam) with GPS and VOC/Temp/Humidity would open up a range of new applications e.g. moving sensor platforms  or keeping track of air quailty in busses and trains. 

Ideas for data structure:
Port 7: 
byte 1-3:	Temp°C
byte 4-5: Humid	
bytes 6-7: VOC
optional as the BME680 does not deliver such data would be  NO2 and PM2.5 / PM10

cyberman54, 2018-11-18:
Will be implemented with v1.6.82 (found currently in development branch).

closed 2018-11-18 by cyberman54

gregorwolf, 2018-11-22:
Could I also use BME280 instead of BME680? Which Pins do I have to connect?

cyberman54, 2018-11-22:
Paxcounter currently relies on the [Adafruit BME680](https://adafruit.github.io/Adafruit_BME680/html/index.html) libary which is specifically designed for the BME680. If you want ot use BME280 you must change this lib, this should'nt be too complicated, just make changes in bme680.cpp 

With next release this probably will be changed to precompiled Bosch BSEC library, which will also work with BME280.

What pins to connect depends on your hardware and on the interface you select, the Bosch BME680 can be used on SPI and I2C. Use the appropriate Arduino pins of your board.