cyberman54/ESP32-Paxcounter #182: Max size of  std::set<uint16_t> macs or life time ?

zoltan85000, 2018-10-11:
Dear,
I would like to know the max number of  hashedmac stored in std::set<uint16_t> macs; 
I don't found in macsniff.cpp and anywhere to flush this set; or some in/out management ?
My question is If a iPhone come near a node, leave the area, and come back in 2 or 3 hours, it's will be count 2 times or 1 ? what is the life of std::set<uint16_t> macs; ?
Best regards.

cyberman54, 2018-10-11:
The lifetime of MACs in the container is controlled.by setting SEND_SECS in paxcounter.conf and can be changes remotely with opcods 0x0A

closed 2018-10-11 by cyberman54

zoltan85000, 2018-10-14:
Dear
so the same MACs can be counted multiple time every SEND_SECS ?
I'm a wrong ?
Best regards

zoltan85000, 2018-10-15:
Dear,
I don't anderstand, how could you assum no double counts after send a payload each 30s  (`SEND_SECS`  in [paxcounter.conf ](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/paxcounter.conf))? 
Since when we are sending payload data by [sendata.cpp ](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/senddata.cpp), `reset_counters()` call `macs.clear()`  in [cyclic.cpp](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/cyclic.cpp)  with `macs.clear(); ` and so we lose all past MACs seen ?
Best regards.

cyberman54, 2018-10-15:
No double counts occur **between two send cycles**

zoltan85000, 2018-10-16:
Dear,
in this example of LOG for me; between two stage of sendata the same MAC is counted twice 👍 I'm 

> [D][macsniff.cpp:122] mac_add(): Added --> Filtered MAC 60:03:08:A5:15:DB
> [D][cyclic.cpp:31] doHousekeeping(): Loraloop 1600 bytes left
> [D][cyclic.cpp:34] doHousekeeping(): Wifiloop 1420 bytes left
> [D][cyclic.cpp:36] doHousekeeping(): Statemachine 428 bytes left
> [I][senddata.cpp:21] SendData(): 4 bytes enqueued to send on LoRa
> [I][senddata.cpp:28] SendData(): 4 bytes enqueued to send on SPI
> [I][senddata.cpp:35] SendData(): Counter cleared
> [I][senddata.cpp:99] checkSendQueues(): 4 bytes sent to SPI
> [D][macsniff.cpp:122] mac_add(): Added --> Filtered MAC 60:03:08:A5:15:DB

I'm a wrong ?
Best regards.

cyberman54, 2018-10-16:
The MAC is stated as filtered, meaning it was not counted, and so it was not necessary to checked if it was known of new. Please read and understand the code, that's why it is open source.