cyberman54/ESP32-Paxcounter #189: Test of development branch

mutton74, 2018-10-22:
Tested this out, thanks for making it available with the band plans.

After following the instructions to setup all the conf files, I have got it working. 

There was an error while building and only worked after commenting out a couple of lines. Below is the error 

Compiling .pioenvs\ttgov1\src\lorawan.cpp.o
src\display.cpp: In function 'void refreshtheDisplay()':
src\display.cpp:148:23: error: 'lora_datarate' was not declared in this scope
u8x8.printf("%c%c", lora_datarate[LMIC.datarate * 2],
^
*** [.pioenvs\ttgov1\src\display.cpp.o] Error 1
*** [.pioenvs\ttgov1\src\display.cpp.o] Error 1 

Also, while a join happens immediately after powering on, the first measurement doesn't happen from anywhere between 5-10 minutes, will have to read the conf file instructions a little more to work that one out. 

Thanks
Paul

cyberman54, 2018-10-22:
Build error should be fixed now, please retry.
Don't know what 's causing the long pause before first transmit, it should happen after the number of seconds set in paxcounter.conf.

mutton74, 2018-10-23:
Retired, works well without error

I suspect the long time to join will be because of sub band choice

In Arduino IDE, in my sketch I usually drop in 

`    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
    LMIC_selectSubBand(1);`

I think I saw the Clcok error setting somewhere but can't see a sub band choice anywhere.

For TTN and AU915 (AU921 in MCCI Library) we need to start at 916.8 - Channel 8 - the second sub band.

cyberman54, 2018-10-23:
subband choice is in the code, please look in lorawan.cpp

cyberman54, 2018-10-23:
I added AU921 support by adding LMIC_selectSubBand(1) in main.cpp line 180 in code in branch development. Could you please re-check and state the results here? Thanks.

mutton74, 2018-10-23:
Have just tested it on a TTGo V2 and Heltec and both work perfectly now, thanks!!!

Joins immediately and then sends data at the specified time, every time.