cyberman54/ESP32-Paxcounter #197: How can you check latest mobile model oui vendors

baggio24881, 2018-11-04:
When you check if an OUI is in vendor array, what happen with the new mobile device models?
How did you get the list of the vendor array of only mobile phones?
And how do you maintain up-to-date this vendor list?
Thanks

spmrider, 2018-11-04:
The check is against reserved vendor ranges as marked in https://standards.ieee.org/develop/regauth/oui/oui.csv
The generation of the vendor list is then done by string matching against known (big) vendors. We try not to go for every small vendor to keep the list in a feasible size for the small device (size and matching effort).
The current code (which in the end needs some manual "fixing") to generate the list is a simple shell script we run on a regular basis to go for new OUIs.

`#!/bin/sh

if [ ! -r "oui.csv" ]
then
        fetch --no-verify-peer https://standards.ieee.org/develop/regauth/oui/oui.csv
fi

size="`egrep -h '(Caterpillar|Motorola Mobility LLC|GUANGDONG OPPO MOBILE TELECOMMUNICATIONS|Huawei Symantec Technologies|Microsoft|HTC Corporation|Samsung Electronics Co.|SAMSUNG ELECTRO-MECHANICS.THAILAND.|BlackBerry RTS|LG ELECTRONICS INC|Apple, Inc.|LG Electronics|OnePlus Tech .Shenzhen. Ltd|Xiaomi Communications Co Ltd|LG Electronics .Mobile Communications.)' oui.csv | wc -l | sed 's/ //g'`"

echo "std::array<uint32_t, ${size}> vendors = {"
egrep -h '(Caterpillar|Motorola Mobility LLC|GUANGDONG OPPO MOBILE TELECOMMUNICATIONS|Huawei Symantec Technologies|Microsoft|HTC Corporation|Samsung Electronics Co.|SAMSUNG ELECTRO-MECHANICS.THAILAND.|BlackBerry RTS|LG ELECTRONICS INC|Apple, Inc.|LG Electronics|OnePlus Tech .Shenzhen. Ltd|Xiaomi Communications Co Ltd|LG Electronics .Mobile Communications.)' oui.csv | cut -d, -f2 | tr '[A-Z]' '[a-z]' | sed 's/^\(.*\)$/  0x\1,/'
echo "  0 };"
`

baggio24881, 2018-11-04:
Ok. Perfectly explained. 
Thanks. You can close this issue/question.
Regards

closed 2018-11-07 by spmrider