cyberman54/ESP32-Paxcounter #206: Cannot compile on Windows 10, Atom, Platformio

pe1mew, 2018-11-21:
Cannot compile this version release of PAXcounter using Windows 10, Atom, Platformio. When compiling this error occurs.

```
src\wifiscan.cpp: In function 'void wifi_sniffer_packet_handler(void*, wifi_promiscuous_pkt_type_t)':
src\wifiscan.cpp:16:9: error: 'wifi_ieee80211_packet_t' does not name a type
const wifi_ieee80211_packet_t *ipkt =
^
src\wifiscan.cpp:18:9: error: 'wifi_ieee80211_mac_hdr_t' does not name a type
const wifi_ieee80211_mac_hdr_t *hdr = &ipkt->hdr;
^
src\wifiscan.cpp:25:24: error: 'hdr' was not declared in this scope
mac_add((uint8_t *)hdr->addr2, ppkt->rx_ctrl.rssi, MAC_SNIFF_WIFI);
^
Compiling .pioenvs\ttgov2\lib666\U8g2_ID942\U8x8lib.cpp.o
*** [.pioenvs\ttgov2\src\wifiscan.cpp.o] Error 1
 [ERROR] Took 19.74 seconds
```
Because the error seems to have i't origin in the compiler not finding the header file wifiscan.h it is included in the include list of platformio.ini to solve the error:
`-include include/wifiscan.h`
As a result of this new errors occur:

```
Compiling .pioenvs\ttgov2\FrameworkArduino\WString.cpp.o
C:\Users\drasv\.platformio\packages\framework-arduinoespressif32\cores\esp32\WMath.cpp: In function 'unsigned int makeWord(unsigned char, unsigned char)':
C:\Users\drasv\.platformio\packages\framework-arduinoespressif32\cores\esp32\WMath.cpp:65:55: error: ambiguating new declaration of 'unsigned int makeWord(unsigned char, u
nsigned char)'
unsigned int makeWord(unsigned char h, unsigned char l)
^
In file included from ./include\hash.h:4:0,
from &lt;command-line&gt;:0:
C:\Users\drasv\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:162:10: note: old declaration 'uint16_t makeWord(byte, byte)'
uint16_t makeWord(byte h, byte l);
^
*** [.pioenvs\ttgov2\FrameworkArduino\WMath.cpp.o] Error 1
[ERROR] Took 71.92 seconds
```
This error is related to a redefinition of function makeWord() in the arduino libraries. Solution is to remove this function from WMath.cpp (my path: C:\Users<username>.platformio\packages\framework-arduinoespressif32\cores\esp32):
```
// unsigned int makeWord(unsigned char h, unsigned char l)
// {
//     return (h << 8) | l;
// }
```
Now the code compiles again :-)

As this DIRTY workaround is no permanent solution someone with more expertise on this platform is requiest to investigate and provide a more permanent solution.

Thanks in advance.

gregorwolf, 2018-11-22:
Same issue here on Mac OSX

cyberman54, 2018-11-25:
@pe1mew @gregorwolf I think this is caused by dependencies of .h files. Still did not find out what triggered this now, but i think i found a quick solution by moving some type definitions from wifiscan.h to wifiscan.cpp. Can you please try again and let me know here?

cyberman54, 2018-11-25:
Should be fixed with current v1.6.9.
If not, please re-open this issue.

closed 2018-11-25 by cyberman54