cyberman54/ESP32-Paxcounter #207: UpdateClass won't compile on case-sensitive filesystem

AmedeeBulle, 2018-11-21:
`src/update.cpp` has:
```
#include "Update.h"
```
But this include does not exists, only `include/update.h` is available.

We should either rename the file in `include/Update.h` or change the `include` statement.

Note that it is a bit unfortunate to keep the same name as the original from the espressif framework, as the compiler doesn't complains about the missing include, but reverts to `.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Update.h` which in turns generates unexpected errors...

AmedeeBulle, 2018-11-21:
Note that `include/ota.h` includes 
```
#include "update.h"
```
so one of these should be fixed...

cyberman54, 2018-11-21:
fixed.

> Note that it is a bit unfortunate to keep the same name as the original from the espressif framework, as >the compiler doesn't complains about the missing include, but reverts to [...]

Next release won't include file update.h any longer, since the necessary patch in update.h will be merged in next arduino-espressif32 core.

This is already done in current version in branch "sensor-development".

closed 2018-11-21 by cyberman54