cyberman54/ESP32-Paxcounter #219: Can't compile - PlatformIO - Ubuntu

kastayou, 2018-12-10:
Using PlatformIO, version 3.6.2 on Ubuntu

After a fresh cloning of the repo and creating/editing the the needed config files:
src/paxcounter.conf
src/lmic_config.h
create file loraconf.h in your local /src directory

I get the following error message during compilation:

```
Compiling .pioenvs/ttgobeam/src/antenna.cpp.o
cc1plus: fatal error: home/xxx/TTGO/PAXCOUNTER/src/hal/ttgobeam.h: No such file or directory
compilation terminated.
Compiling .pioenvs/ttgobeam/src/battery.cpp.o
cc1plus: fatal error: home/xxx/TTGO/PAXCOUNTER/src/hal/ttgobeam.h: No such file or directory
compilation terminated.
Compiling .pioenvs/ttgobeam/src/blecsan.cpp.o
cc1plus: fatal error: home/xxx/TTGO/PAXCOUNTER/src/hal/ttgobeam.h: No such file or directory
compilation terminated.
```

To solve the problem I had to modify the platformio.ini file

from:
```
build_flags_basic =
    -include $PROJECTSRC_DIR/hal/${PIOENV}.h
    -include $PROJECTSRC_DIR/paxcounter.conf
```
to:
```
build_flags_basic =
    -include **/**$PROJECTSRC_DIR/hal/${PIOENV}.h
    -include **/**$PROJECTSRC_DIR/paxcounter.conf
```
After this modification, the code compiles correctly

cyberman54, 2018-12-11:
Thanks for reporting this bug.
I tested on Windows10 platform. Unfortunately with additional "/" it does not compile on windows.
Looks like a platform dependent problem.
I will open an issue at platformio.

cyberman54, 2018-12-11:
Not a platformio bug.
Path in platformio.ini was invalid.
Fixed it.

kastayou, 2018-12-11:
I am using Visual Studio Code as editor.

as you can see in my initial message with the following configuration in the platformio.ini file: 
```
build_flags_basic =
    -include $PROJECTSRC_DIR/hal/${PIOENV}.h

```
I had the following error message:
```
cc1plus: fatal error: home/xxx/TTGO/PAXCOUNTER/src/hal/ttgobeam.h: No such file or directory
compilation terminated.

```
The 'src' directory is there even if not present in the build_flag_basic because the variable $PROJECTSRC_DIR is used

With the modification done:
```
build_flags_basic =
    -include $PROJECT_DIR/src/hal/${PIOENV}.h
```

I have the same error message.
I guess that $PROJECTSRC_DIR = $PROJECT_DIR/src

But if I do the same modification
    -include $PROJECT_DIR/src/hal/${PIOENV}.h
=>
    -include /$PROJECT_DIR/src/hal/${PIOENV}.h

I have no error during the compilation

In the previous ini file that I was using, the ini file was:
```
build_flags_basic =
    -include "src/hal/${PIOENV}.h"
    -include "src/paxcounter.conf"

```
In that case, it compiles also without error.

cyberman54, 2018-12-11:
Allright, i didn't see that you changed the variable in platformio.ini.
The point is that, regardless if using $PROJECT_DIR or $PROJECTSRC_DIR, compilation on Windows systems (at least with VSC) breaks, if we add the "/" at the beginning of the path.

So we need a solution here that works platform independent.

I reopen this.

kastayou, 2018-12-11:
In the previous version (1.6.5x), the ini file was:
```
build_flags_basic =
    -include "src/hal/${PIOENV}.h"
    -include "src/paxcounter.conf"

```
It is working fine on ubuntu but I cannot test it on windows. It might be the solution that is platform independant.

cyberman54, 2018-12-11:
I tested on Windows, it works.
Don't remember why/how this change came in.
I reverted to previous version as suggested by you.
Thanks again for working on this issue.

closed 2018-12-11 by cyberman54