cyberman54/ESP32-Paxcounter #222: TTGO Beam stopps sending after join and one message

sbamueller, 2018-12-16:
[image]
[image]
I flashed the TTGO Beam. When I power it up it produces a join and then sends one message with BLE/WIFI. Then the device stops sending. No matter if I send 0301 or any other message. It stopps sending. I'm not sure what I can do anymore. 
I have spoken with Bahnhofsoma (https://twitter.com/Bahnhofsoma/status/1074008037600452608) and also with a guy at TTN Stuttgart Meeting who both have the same problem.

cyberman54, 2018-12-16:
Please increase debug level to 4 in platformio.ini, then flash again and monitor output on serial console. Please post output here.

cyberman54, 2018-12-16:
try changing version in platformio.ini from 1.6.971 to 1.6.972, reflash and report the result here.

cyberman54, 2018-12-17:
@sbamueller ask the supplier.
"T22_V07" is probably the newer version. So try ttgobeam_new.h

cyberman54, 2018-12-17:
Thanks, so it close this issue.
If problems persist, please re-open this issue.

closed 2018-12-17 by cyberman54

sbamueller, 2018-12-17:
Tjakkaaaaa works