cyberman54/ESP32-Paxcounter #225: OTAA JOIN not working with Objenious

kastayou, 2018-12-19:
I had no issue joining with The Things Network using my own gateway but joining the Objenious network (French provider) was difficult.
Difficult means that I have to wait several hours before the OTAA join accept sent by the network provider is accepted by my TTGO T-beam node.

I could see on the Objenious console that the Join request was received and accepted (no credential issue).

I have noticed that by modifying the following line in the lmic_config.h 

`#define CLOCK_ERROR_PROCENTAGE 20
`
=>
`#define CLOCK_ERROR_PROCENTAGE 3
`
The join accept is more easily accepted by the node.

Due to the poor network coverage in my area, the join request reaches the network only at SF11 or SF12

This could help others with similar issue

cyberman54, 2018-12-19:
The parameter CLOCK_ERROR_PROCENTAGE stretches the LoRaWAN receive window, what means that a node has a greater chance to receive data from the gateway (downlink) it it's clock has no precise timing. This means the rx radio is switched on earlier and switched off later. This sure will have impact on power consumption, but i don't see how drecreasing this value could affect OTA performance.

cyberman54, 2018-12-19:
@kastayou can you please check which downlink windows (RX1 oder RX2) your LoRaWAN network provider is using fot OTA?

closed 2019-02-04 by cyberman54