cyberman54/ESP32-Paxcounter #226: Button

qvs5010, 2018-12-20:
Hi

I want to ask, how can i connect a external button.

I want to connect the following button.

[https://images-na.ssl-images-amazon.com/images/I/61Tv47rW6cL._SX679_.jpg](https://images-na.ssl-images-amazon.com/images/I/61Tv47rW6cL._SX679_.jpg)

It has 2 wires, so my understanding is, i can connect one wire to GND (Ground) and the other to a GIOP Pin 22?

And then just change the GIOP ping in the ESP32-Paxcounter/src/hal/ttgobeam.h from 

#define HAS_BUTTON GPIO_NUM_39 // on board button "BOOT" (next to reset button)

to

#define HAS_BUTTON GPIO_NUM_22 // on board button "BOOT" (next to reset button)

But do i need to connect VCC anywhere, and do i have to use a resistor and how big if i need to?

Thanks
Q

cyberman54, 2018-12-20:
It should work this simple way
 Have in mind that you have no button debouncing this way, but can be done by software if.needed.

qvs5010, 2018-12-20:
Hi Cyberman

Thanks

But must i connect the VCC with a resistor?

like the following.

[https://lastminuteengineers.com/wp-content/uploads/2018/11/Connecting-Button-to-ESP32-for-ext0-External-Wakeup-Source.png](https://lastminuteengineers.com/wp-content/uploads/2018/11/Connecting-Button-to-ESP32-for-ext0-External-Wakeup-Source.png)

cyberman54, 2018-12-20:
No.

qvs5010, 2018-12-20:
Aaaa cool, so it just GIOP 22 to Ground.

Thanks

closed 2018-12-20 by cyberman54