cyberman54/ESP32-Paxcounter #231: BME680 and DISPLAY not working parallel

cyberman54, 2018-12-25:
BME680 sensor and OLED display both use i2c communication from separate parallel tasks.
Since esp32 i2c hal is not thread safe, we need a mutex solution for this.

cyberman54, 2018-12-27:
fixed in v1.7.03 (currently in development-bme680-2 branch)

closed 2018-12-27 by cyberman54

mcapani, 2021-03-12:
good evening, I know that you have closed the trend, it is days that I have read everything, but I still have this problem, I have the ttgo t3 v 2.1 with display and a sensor bme 680. I have tried the various versions but I always have the probklema that together they don't work.

"Starting I2C bus scan ...
23: 06: 23.191> [I] [i2c.cpp: 78] i2c_scan (): 0 I2C device (s) found
23: 06: 23.494> [E] [display.cpp: 70] dp_setup (): FAILURE in src / display.cpp: 70
23: 06: 23.494> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4826 (xTaskNotify) - assert failed! "
can you please give me some advice ???
thank you

cyberman54, 2021-03-12:
Please show your `platformio.ini` and `paxcounter.conf`

mcapani, 2021-03-12:
; PlatformIO Project Configuration File
; NOTE: PlatformIO v5 is needed!
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

; ---> SELECT THE TARGET PLATFORM HERE! <---
[board]
;halfile = generic.h
;halfile = ebox.h
;halfile = eboxtube.h
;halfile = ecopower.h
;halfile = heltec.h
;halfile = heltecv2.h
;halfile = heltecv21.h
;halfile = ttgov1.h
;halfile = ttgov2.h
;halfile = ttgov21old.h
halfile = ttgov21new.h
;halfile = ttgofox.h
;halfile = ttgobeam.h
;halfile = ttgobeam10.h
;halfile = ttgotdisplay.h
;halfile = ttgotwristband.h
;halfile = fipy.h
;halfile = lopy.h
;halfile = lopy4.h
;halfile = lolin32litelora.h
;halfile = lolin32lora.h
;halfile = lolin32lite.h
;halfile = wemos32oled.h
;halfile = wemos32matrix.h
;halfile = octopus32.h
;halfile = tinypico.h
;halfile = tinypicomatrix.h
;halfile = m5core.h
;halfile = m5fire.h
;halfile = olimexpoeiso.h

[platformio]
; upload firmware to board with usb cable
default_envs = usb
; upload firmware to a jfrog bintray repository
;default_envs = ota
; use latest versions of libraries
;default_envs = dev
description = Paxcounter is a device for metering passenger flows in realtime. It counts how many mobile devices are around.

[common]
; for release_version use max. 10 chars total, use any decimal format like "a.b.c"
release_version = 2.4.0
; DEBUG LEVEL: For production run set to 0, otherwise device will leak RAM while running!
; 0=None, 1=Error, 2=Warn, 3=Info, 4=Debug, 5=Verbose
debug_level = 3
extra_scripts = pre:build.py
otakeyfile = ota.conf
lorakeyfile = loraconf.h
lmicconfigfile = lmic_config.h
platform_espressif32 = espressif32@3.1.0
monitor_speed = 115200
upload_speed = 115200 ; set by build.py and taken from hal file
display_library = ; set by build.py and taken from hal file
lib_deps_lora =
    mcci-catena/MCCI LoRaWAN LMIC library @ ^3.3.0
lib_deps_display =
    bitbank2/OneBitDisplay @ 1.9.0
    bitbank2/BitBang_I2C @ ^2.1.3
    ricmoo/QRCode @ ^0.0.1
    bodmer/TFT_eSPI @ ^2.3.51
lib_deps_ledmatrix =
[... 184 lines cut ...]
#define TIMEPORT                        9       // time query and response
#define SENSOR1PORT                     10      // user sensor #1
#define SENSOR2PORT                     11      // user sensor #2
#define SENSOR3PORT                     12      // user sensor #3

// Cayenne LPP Ports, see https://community.mydevices.com/t/cayenne-lpp-2-0/7510
#define CAYENNE_LPP1                    1       // dynamic sensor payload (LPP 1.0)
#define CAYENNE_LPP2                    2       // packed sensor payload (LPP 2.0)
#define CAYENNE_GPS                     3       // full scale GPS payload
#define CAYENNE_ACTUATOR                10	    // actuator commands
#define CAYENNE_DEVICECONFIG            11	    // device period configuration
#define CAYENNE_SENSORREAD              13	    // sensor period configuration
#define CAYENNE_SENSORENABLE            14	    // sensor enable configuration

// MQTT settings, only needed if MQTT is used (#define HAS_MQTT in board hal file)
#define MQTT_ETHERNET 0 // select PHY: set 0 for Wifi, 1 for ethernet
#define MQTT_INTOPIC "paxin"
#define MQTT_OUTTOPIC "paxout"
#define MQTT_PORT 1883
#define MQTT_SERVER "public.cloud.shiftr.io"
#define MQTT_USER "public"
#define MQTT_PASSWD "public"
#define MQTT_RETRYSEC 20  // retry reconnect every 20 seconds
#define MQTT_KEEPALIVE 10 // keep alive interval in seconds
//#define MQTT_CLIENTNAME "my_paxcounter" // generated by default

cyberman54, 2021-03-12:
@mcapani You selected ttgov21new.h as board hal file. Are you sure you have this board? Please show a photo of your board to verify version is matching the selected hal file.

mcapani, 2021-03-12:
[image]
[image]

cyberman54, 2021-03-13:
Which pins did you use to connect the BME680?

mcapani, 2021-03-13:
I had the doubt .... in some guides I read that the module I purchased (photo1) can also be powered at 5v because the board adjusts the voltage automatically. in any case I connected 2 photos as shown in the attached diagram. 21/22 5v gdn
[image]
[image]

cyberman54, 2021-03-13:
Please show your hal file. Did you add/edit the BME680 settings in hal file?

mcapani, 2021-03-13:
now works!!!! I'm a bit ashamed :) you have hit the problem !! great i did a lot of tests and it was trivially a 3.3V pin soldering error. I apologize for having lost this time, probably yesterday I was "melted". Thanks again. now I do the other tests and maybe we will talk about something else ..... ps it also works to detect immune apps (covid-italia) thanks

cyberman54, 2021-03-13:
If you need support, please use discussion tab next time, not issue. Thx.

mcapani, 2021-03-13:
ok thanks