cyberman54/ESP32-Paxcounter #238: Display doesnt work on T-Beam

theArcher73, 2019-01-04:
with Version 1.6.972 the display worked well, but with the current version there is no function.
added in the 
platform.io / [env:ttgobeam_new] / lib_deps the line 	${common.lib_deps_display}

and in the ttgobeam_new.h the following rows

#define HAS_DISPLAY U8X8_SSD1306_128X64_NONAME_HW_I2C
//#define DISPLAY_FLIP  1 // uncomment this for rotated display
// Hardware pin definitions for TTGO V2 Board with OLED SSD1306 0,96" I2C Display
#define OLED_RST U8X8_PIN_NONE // connected to CPU RST/EN
#define OLED_SDA GPIO_NUM_21  // ESP32 GPIO21 – SD1306 D1+D2
#define OLED_SCL GPIO_NUM_22 // ESP32 GPIO22 – SD1306 D0

Is there just an other File to update?

theArcher73, 2019-01-04:
I use an TTGO T-Beam T22V07

cyberman54, 2019-01-04:
Allright, here we go: You have wrong pin definitions in hal file, please use "MY_":

#define MY_OLED_SDA (21)
#define MY_OLED_SCL (22)
#define MY_OLED_RST U8X8_PIN_NONE

theArcher73, 2019-01-04:
thx a lot, now the display works fine. The last Problem is that there is no GPS-Fix, concret 0 Sat's in the display....