cyberman54/ESP32-Paxcounter #239: OTAA not joining with TTBEAM

MartinLocher, 2019-01-06:
Dear all,

unfortunately my ttbeam (new) is not joing with OTAA on thethingsnetwork. I guess it is related to an timing issue as I tried with an "normal" ardunio sketch for the ttbeam and this works perfect. Same settings and i tried several lsb msb options for the keys. ttnnetwork is getting the same information however it doesnt commence. Also i tried to set different clockerrror settings without luck. However, this is a great piece of craftmanship on how to code. Chapeau !

Best regards

cyberman54, 2019-01-06:
Please give exact information:
- TTGO-T-Beam version (what is printed on the pcb?)
- Country & LoRaWAN settings used
- Which paxcounter software version did you use?
- Which "normal" arduino sketch did you use
- Do you have access to a TTN gateway and can you look in the gateway's log, if you see join attempts of the device?

MartinLocher, 2019-01-07:
Thanks for the quick answer. Please find the info below.

TTGO-T-Beam version (what is printed on the pcb?): T22_V07 20180711 / I use the build option ttgobeam_new
Country & LoRaWAN settings used: #define CFG_eu868 1
Which paxcounter software version did you use?: The latest
Which "normal" arduino sketch did you use:
But i use there a different lmic lib !!!
/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in config.h.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8]={ same keys };
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ same keys };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
// The key shown here is the semtech default key.
static const u1_t PROGMEM APPKEY[16] = { same keys };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;
#define LoRa_RST  14  // GPIO 14
#define LoRa_CS   18  // GPIO 18
#define LoRa_DIO0 26  // GPIO 26
#define LoRa_DIO1 33  // GPIO 33
[... 96 lines cut ...]
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}

Do you have access to a TTN gateway and can you look in the gateway's log, if you see join attempts of the device?, Yes I have access which log do you want syslog ?

I see in the console the joining attempt but then it stopps.

Thanks again for your support !!!
All the best Martin

MartinLocher, 2019-01-07:
Just an idea ... can i "just" change the libs ? I am not yet familiar with platform.io so i would need your help here. as far as i can see platform.io is automatically downloading the needed libs and i have no glue how to intersect ... :-)

cyberman54, 2019-01-07:
I can confirm that the above setup (TTGO T-Beam T22_V07 on TTN with EU parameters) is working, i tested it.

If you see on the LoRaWAN gateway only a single join request, that does not repeat after 60 seconds, your TTGO probably crashes after join request.

1)
Check your USB power. To ensure that it is not an USB power issue, operate the TTGO on a fully charged 18650 cell. Disable brownout detection by enabling this line in ttgobeam_new.h:
`#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature`

2)
Increase application debug level in platformio.ini to 4: `debug_level = 4`
Increase lmic debug level in lmic_config.h to 2: `#define LMIC_DEBUG_LEVEL 2`
Rebuild software and flash. Then write serial log on console and check contents

3)
If you don't see the root cause post the console log here.

MartinLocher, 2019-01-07:
The TTGO is running not on Battery just on the usb connected to a mac. 

I see single join requests after 2 minutes (19:32, 19:34)

Here is the log:
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 1
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 2
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 1
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 2
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][cyclic.cpp:37] doHousekeeping(): Wifiloop 576 bytes left
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 720 bytes left
[D][cyclic.cpp:41] doHousekeeping(): Gpsloop 1528 bytes left
[D][cyclic.cpp:49] doHousekeeping(): LEDloop 552 bytes left
[D][battery.cpp:51] read_voltage(): Raw: 2511 / Voltage: 4226mV
[I][cyclic.cpp:55] doHousekeeping(): Measured Voltage: 4226mV
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 1
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 2
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 1
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 2
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[... 935 lines cut ...]
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][senddata.cpp:70] sendCounter(): No valid GPS position
[I][lorawan.cpp:413] lora_enqueuedata(): 4 bytes enqueued for LORA interface
[I][senddata.cpp:49] sendCounter(): Counter cleared
[D][battery.cpp:51] read_voltage(): Raw: 2511 / Voltage: 4226mV
[I][lorawan.cpp:413] lora_enqueuedata(): 2 bytes enqueued for LORA interface
[D][cyclic.cpp:37] doHousekeeping(): Wifiloop 544 bytes left
[D][cyclic.cpp:39] doHousekeeping(): IRQhandler 448 bytes left
[D][cyclic.cpp:41] doHousekeeping(): Gpsloop 1528 bytes left
[D][cyclic.cpp:49] doHousekeeping(): LEDloop 552 bytes left
[D][battery.cpp:51] read_voltage(): Raw: 2504 / Voltage: 4216mV
[I][cyclic.cpp:55] doHousekeeping(): Measured Voltage: 4el 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 1
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 2
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7

cyberman54, 2019-01-07:
The log is okay.
The device did either not receive or did not capture the join accept message from the gateway.

Possible reasons are:
1) There was no join accept message sent by the gateway (check your gateway logs)
2) join accept message was not received
3) join accept message was missed due to a timing issue

ad 1): you're using a predefined DEVEUI from your loraconf.h. Double check that this DEVEUI is registered for the used APPEUI.

ad 2):  Make sure your device is in range of gateway. You may wait up to 1 hour. LMIC retries and increases SF level during join, starting with SF=7.

ad 3): check your TTN setup with other nodes

For reference you find enclosed a log from my TTGO T-Beam which succesfully joins TTN with the same code.

---

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[I][main.cpp:94] setup(): Starting PAXCNT v1.6.972
[I][main.cpp:107] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
[I][main.cpp:108] setup(): ESP32 SDK: v3.2-dev-39-gaaf12390
[I][main.cpp:109] setup(): Free RAM: 4345112 bytes
[I][main.cpp:112] setup(): TinyGPS+ v1.0.2
[I][configmanager.cpp:177] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:50] open_storage(): Opening NVS
[I][configmanager.cpp:55] open_storage(): Done
[I][configmanager.cpp:191] loadConfig(): NVRAM settings version = 1.6.972
[I][configmanager.cpp:208] loadConfig(): lorasf = 9
[I][configmanager.cpp:216] loadConfig(): txpower = 15
[I][configmanager.cpp:224] loadConfig(): adrmode = 1
[I][configmanager.cpp:232] loadConfig(): screensaver = 0
[I][configmanager.cpp:240] loadConfig(): screenon = 1
[I][configmanager.cpp:248] loadConfig(): countermode = 0
[I][configmanager.cpp:256] loadConfig(): sendcycle = 30
[I][configmanager.cpp:264] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:272] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:280] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:288] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:296] loadConfig(): blescantime = 8
[I][configmanager.cpp:304] loadConfig(): BLEscanmode = 1
[I][configmanager.cpp:312] loadConfig(): rssilimit = 0
[I][configmanager.cpp:320] loadConfig(): payloadmask = 4294967295
[I][configmanager.cpp:328] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:336] loadConfig(): Run mode = 0
[I][configmanager.cpp:343] loadConfig(): Done
[I][main.cpp:134] setup(): Starting LEDloop...
[I][battery.cpp:30] calibrate_voltage(): ADC characterization based on reference voltage stored in eFuse
[D][battery.cpp:51] read_voltage(): Raw: 2450 / Voltage: 4256mV
[I][gpsread.cpp:23] gps_init(): Using serial GPS
[I][main.cpp:193] setup(): Starting GPSloop...
[I][lorawan.cpp:374] lora_stack_init(): LORA send queue created, size 510 Bytes
[I][lorawan.cpp:376] lora_stack_init(): Starting LMIC...
RXMODE_RSSI
[I][main.cpp:284] setup(): Features: LED BLE BATT BTN_PD GPS LORA OUIFLT PACKED
[I][macsniff.cpp:36] printKey(): DevEUI <...>
[I][macsniff.cpp:36] printKey(): AppEUI: <...>
[I][macsniff.cpp:36] printKey(): AppKey: <...>
[I][main.cpp:296] setup(): Starting Bluetooth...
[I][blecsan.cpp:247] start_BLEscan(): Initializing bluetooth scanner ...
[... 43 lines cut ...]
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 3
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 4
647796: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 4
647896: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
660628: Setup channel, idx=3, freq=867100000
660638: Setup channel, idx=4, freq=867300000
660643: Setup channel, idx=5, freq=867500000
660689: Setup channel, idx=6, freq=867700000
660939: Setup channel, idx=7, freq=867900000
[I][lorawan.cpp:209] onEvent(): ADR=1, SF=9, TXPOWER=15
[I][lorawan.cpp:285] onEvent(): EV_JOINED
661725: engineUpdate, opmode=0x800
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 5
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 6
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 7
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 8
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 9
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 10
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 11
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 12
[D][wifiscan.cpp:72] switchWifiChannel(): Wifi set channel 13
```

---

MartinLocher, 2019-01-07:
I am missing the EV_JOINED. I think i give up here as it doesnt work with a ttgov2 neither ... i see the paxounter splash on the lcd waiting, waiting for a join ...

cyberman54, 2019-01-07:
You may try to set DEVEUI in loraconf.h to zero, meaning that DEVEUI is generated by the TTGO and printed on the console during startup. Use this DEVEUI with copy/paste to create a fresh device on TTN console and try again.

I can't help you further with this.

I think it is not a paxcounter related issue, so i close this now.

If you find a solution, please comment here, or re-open.

closed 2019-01-07 by cyberman54

MartinLocher, 2019-01-08:
Hi cyberman,
it works now ... 
I am so sorry that I wasted your time. I think i fiddled too much around with the code once it was not working ?! Because i didnt saw any console message in ttn coming through but i dont want to search excuses. It was my bad !
I downloaded again your code, which is for me again a real real excellent example how to code on a mulitiproc and it worked out of the box. 

 I owe you at least 2 beer. 

Will you join lorawan conference in Amsterdam ? That would be the perfect time to have a few drinks on me !

Thank you again and best regards, Martin

cyberman54, 2019-01-08:
@MartinLocher Yes, you can serve me at least 2 bear at TTN conference, i will join.

Don't overestimate the quality of the code. It's far away from an excellent level, there are lot's of design flaws. The code is on a beginner level. 

Goal of project was not to give a coding framework, but to show an interesting iot use case in terms of use/value and cost, and to put this open source, for encouraging senior code developers to grow and refactor the whole project. But we're still behind this goal.

aapris, 2019-01-09:
I could serve a beer too. :) Where are you going to drink those and which day?

cyberman54, 2019-01-11:
@martinlocher @aapris there will be an offical talk on paxcounter by @spmrider in the ttn conf program. We will try to organize a meetup for attendees of this talk after.

MartinLocher, 2019-01-27:
Cyberman54 in which slot of the conference is the talk ?

cyberman54, 2019-01-27:
Yey, Day2, Stage1, 3:00 PM

MartinLocher, 2019-01-27:
Damned, i am not sure if we can make it as our flights are leaving early. we are in the nearby Hilton, perhaps we can meat on Wednesday or Thursday ?