cyberman54/ESP32-Paxcounter #248: add information about the connections for the bme 680

sbamueller, 2019-01-20:
Please add:

Connecting a BME 680 ist done easily, this way the PaxCounter can also transmit data about the Air Quality, such as temperature, humditiy, pressure and VOC.

If you use the BlueDot sensor connect as follows:

Blue Dot --> TTGO
SDI --> SDA (Pin 21)
SCK --> SCL (Pin 22)
VCC --> 3V3
GND --> GND

When connecting to the TTGO Board change the following in the software:

File: src/hal/ttgobeam_new.h
uncomment Programline 16: 
#define HAS_BME GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL

If you use Bluedot the default Adress in the I2C bus is: 0x77

PaxCounter uses the standard Bosch Library:

/** BME680 I2C addresses */
#define BME680_I2C_ADDR_PRIMARY	UINT8_C(0x76)
#define BME680_I2C_ADDR_SECONDARY	UINT8_C(0x77)

So you have to change the file to:
/** BME680 I2C addresses */
#define BME680_I2C_ADDR_PRIMARY	UINT8_C(0x77)
#define BME680_I2C_ADDR_SECONDARY	UINT8_C(0x76)

cyberman54, 2019-01-20:
Address of BME680 must be selected by hardware, not by software. This is done by wiring the SDO pin. According to BOSCH BME680 this pin *must* be wired, otherwise address of BME680 is undefined:

_Connecting SDO to GND results in slave address 1110110 (0x76); connection it to VDDIO results in slave
address 1110111 (0x77), which is the same as BMP280’s I²C address. The SDO pin cannot be left floating; if left floating, the I²C address will be undefined._

cyberman54, 2019-01-20:
This is already noted in the .h file

//#define BME_ADDR BME680_I2C_ADDR_PRIMARY // connect SDIO of BME680 to GND

so i close this.

closed 2019-01-20 by cyberman54