cyberman54/ESP32-Paxcounter #250: Style: replace #ifdef FOO by #if(FOO)

spmrider, 2019-01-20:
I suggest to replace occurences of #ifdef FOO for all boolean checks by #if(FOO)

This enables to switch features by defining a value true or false (1 or 0 if you prefer), but would still be compatible with the current comment in/out way of configuring things. The master #define FOO could also be surrounded by #ifndef FOO - this would open the config logic to a local config file that either sets some values with compile option or a local config (gitignore'd) config.

Better ideas?

cyberman54, 2019-01-20:
@spmrider Please create new branch based on current development branch to try this.

spmrider, 2019-03-11:
In branch ifdef-dev I replaced several #ifdev / #if defined by #if(...)
Before opening the pull request could someone else check and test, please?

cyberman54, 2019-03-13:
Merged with branch development, no compile issues so far.

cyberman54, 2019-03-15:
done.

closed 2019-03-15 by cyberman54