cyberman54/ESP32-Paxcounter #256: Version 1.7.14 broken for non RTC

spmrider, 2019-01-28:
compiling 1.7.14 (from master) raises two compile errors in cyclic.cpp and lorawan.cpp where the #endif HAS_RTC statement is too early and thus myTZ is needed, though rtctime.h was not included. I moved the #endif after the log message and the code compiled and ran fine.

spmrider, 2019-01-28:
Fix has been committed to master

cyberman54, 2019-01-28:
fixed.

closed 2019-01-28 by cyberman54

aapris, 2019-01-28:
Sorry to comment closed issue, but I'm wondering why time zones are supported at all? I've found (during last 15 years I've been playing around sensor data) that when ever I see "local time" or "time zone" mentioned, it leads almost every time into a mess. I'd say keep all devices and databases in UTC and convert the timestamps to user's local time in user interface or when creating reports etc.

cyberman54, 2019-01-28:
@aapris That's right, and we will keep UTC on the paxcounter for all control tasks. But for the use case "IF482/DCF77 clock controller" we need to provide local time by the sensor. That's why i integrated time zone support. The pain to do it was low since usage of libs.