cyberman54/ESP32-Paxcounter #264: Enhancement - Show number of new PAX since last data transmission

dakoller, 2019-02-05:
In flow meter scenarios it would be helpful to understand, how many signals have already been there during the last measurement, and how many signals are new.

This would help to understand at the location of the paxcounter what the dynamics of pax is... same device all the time OR a big crowd of people, which are wlaking through a street.

For that paxcounter would need to keep the last of the last time recognized networks/devices and process it when the next measurement is due.

cyberman54, 2019-02-05:
This function would force the code to re-identify devices (= users) on some way, what would probably not GDPR / DSGVO compliant.

Thus i don't see this function on the future paxcounter roadmap.

If you know a way how to implement this without privacy issues, please let us know here. We can then discuss how to bring this in the code.

dakoller, 2019-02-05:
assumption is/was that you'd be able to identify the individual e.g. wlan signals  ( without any link to a person... question is just whether the same signal was already present during the last measurements)

I'll check the code on that,

cyberman54, 2019-02-05:
We don't measure Wifi signals on a PHY level, we decode their payload = MAC address. And we don't want to persistently store that, neither the timestamps when and not then location where they were seen.

Goal of paxcounter software is to anonymize sniffed MACs as soon and as safe as possible.

So your approach won't work.

closed 2019-02-06 by dakoller