cyberman54/ESP32-Paxcounter #265: Add SDS011 support

sbamueller, 2019-02-06:
Why not SDS011 support for the PaxCounter?
We could measure the Airquality in relation to the amount of people in a room. Even Control how good the air on the station is. Already the PaxCounter has a BME680 support.

aapris, 2019-02-07:
Do humans emit particulate matter (PM10/PM2.5)? I'd say not so much. Maybe when people move around, they may raise some dust from the floor. The life time of SDS011 is max one year (in continuous use) so it should be replaceable every now and then.

spmrider, 2019-02-07:
@aapris Humans sometimes emit particulate matter, especially the ones smoking. It might be interesting to detect smokers this way. Also pm is on the other hand something that has an effect on humans. So it might be interesting in some scenarios to react, if there are (many) people and the pm values rise.
@sbamueller The SDS011 / SDS018 has good availability is pretty cheap, so it's good for home projects. It has limited life time (as the previous poster states and I already tried for myself) in continouus operation. This can be prolongued by operation in measurement intervals. However, the Sensirion SPS30 is available (at a little higher cost), but seems well more longlived. Also the SPS30 features a wider range of acceptable humidity, which for outdoor operations must be preferred.

sbamueller, 2019-02-07:
I’m open about the specific model. But I consider PM 10 or 2.5 value something interesting for public transport as well. Eg has ventilation on stations to be increased if the air quality is bad, or do people not gather in specific places if the air is bad. 
Are there people smoking in places they should not be.

AugustQu, 2019-12-06:
using the sensor SDS011: I can do this. 4 cables plus a little bit of code - no problem.

For this sensor I have a mobile system which I used and reported about here:
https://linux-fuer-wi.blogspot.com/2019/02/feinstaub-in-wiesbaden-erbenheim.html
https://linux-fuer-wi.blogspot.com/2019/03/feinstaub-in-wiesbaden.html
and naturally:
https://linux-fuer-wi.blogspot.com/2019/01/diesel-verbot-silvester.html

the code for this system was developed in cooperation with CodeForFrankfurt:
https://codeforfrankfurt.github.io/hackathon2018/index.html

AugustQu, 2019-12-28:
I will start looking for this.

AugustQu, 2020-01-09:
a short interim report on this:

this is the board i am using: https://de.aliexpress.com/item/32915894264.html

Finally I managed to get the sensor working with this board, software used is a standalone one.

I will now look integrating my software into the PaxCounter-code and then report here.

cyberman54, 2020-01-10:
@AugustQu SDS011 has a serial interface feeding data with 1 Hz. Thus, it must be polled, similary to NMEA serial GPS. We need a serial buffer and either a 1) separate feeder task or 2) a cyclic interrupt in paxcounter code to keep the polling non blocking. 

1) see gpsread.cpp / 2) see bmesensor.cpp for examples how to integrate polling in paxcounter code.

AugustQu, 2020-01-18:
Hi,

I have finished the version 0.1 of this code. I will download the latest version of PaxCounter from here and implement my modifications on a fresh copy of the code. After this step I want to upload my code here. Where can I do this?

cyberman54, 2020-01-18:
@AugustQu I opened a [new branch SDS011](https://github.com/cyberman54/ESP32-Paxcounter/tree/SDS011). Please make a pull request for this branch, thanks.

AugustQu, 2020-01-21:
OK, I added my modifications.

Unfortunately I made some mistakes:

- I created multiple pull-requests
- 2 of them go to the wrong branch: master

please excuse.

This is the version 0.1 of this software. Please check it. We can continue our discussion here.

AugustQu, 2020-01-21:
some remarks:
in the file hal/ttgov21new.h (= my board) I added this line:
`#define HAS_SDS011  1      // we use the SDS011 connected to this board`

In the case of your board you have to add this line in the corresponding hal-file.

And please look into the file include/sds011read.h. Check these line and modify them for your board:
`
// used pins on the ESP-side:
#define ESP_PIN_TX 19     // connect to RX on the SDS011
#define ESP_PIN_RX 23     // connect to TX on the SDS011
`

I hope that's all for today.

AugustQu, 2020-01-21:
forgotten:
the code now "only" writes the data for PM10 and PM2.5 to the sd-card.

Is it useful to transmit this data via LoRaWAN? I think so, so I will look into the code. But this means, that the script on the TTN-side has to be modified too (and maybe more....)

cyberman54, 2020-01-21:
@AugustQu I merged the PRs in branch SDS011, except two PRs that addressedthe  master branch; please reopen these two PRs for branch SDS011.

AugustQu, 2020-01-22:
Hi,

I updated the 2 pullrequest that went wrong. Plus I added another pullrequest  with some code for sending the data via LoRaWAN. Hope I did it correctly now.

In the moment I have some trouble with a connection to alocal  LoRaWAN-gateway here so I cannot guarantee that it's working correctly. But it did work here in 2 of my tests.

Can you please look into the code so that we can start our discussion.

In the file src/payload.cpp you will find some new member-functions. This is a first shot at it, we can discuss the format of the data transmitted. And also of all possible values of PAYLOAD_ENCODER only 2 are implemented yet.

AugustQu, 2020-01-22:
an observation:
in src/sds011read.cpp the function sds011_loop() contains this line:
`   int sdsErrorCode = sdsSensor.read(&pm25, &pm10);
`
it looks like it returns 0 on error and 1 on success. I have to check the manual for this.

a question:
The SDS-sensor draws between 80 and 100 mA. I can reduce this by putting the sensor to sleep, wake it up (and wait 30 seconds) and then read the values. Is this useful?

For this I need an interrupt-routine that will be called every 30 seconds. Is there something like this already?

AugustQu, 2020-01-22:
OK, checked the source code:
the read()-function of the lib SDS011 returns 0 on success and 1 on error. I will correct my source-code here.

The code in the current version delivers values, so I do not hurry.

cyberman54, 2020-01-22:
@AugustQu I peeked in the code. Currently the sensor values are read inside senddata(). This is a very limited approach, since we cannot have averaging of timeseries and sleep/standby management. Better approach is (as i stated above) to read the sensor data

1) either continously in a separate task,
2) or interrupt driven by an interrupt handler

You find examples for both approaches in paxcounter code, for 1) look in gpsread.cpp, for 2) look in bmesensor.cpp.

Which approach to be chosen depens on how many data and how fast shall be transferred between SDS sensor and cpu.

spmrider, 2020-01-23:
Putting the sensor to sleep and just waking it up every 10min or so for 
30secs for the measurement is absolutely useful. In continuous mode the 
lifetime of the sensor is just 1-1.5 years.

AugustQu, 2020-01-23:
@cyberman54 this is the first shot at this problem. And I will look into the files you mentioned.

AugustQu, 2020-01-23:
@spmrider how many data do you want? how often shall the values be reported?

Currently the software reports the values every minute via LoRaWAN and writes these values to the SD-card. When you want to report fine-dust-values only 1 in 10 minutes then this will be a different approach.

AugustQu, 2020-02-01:
Hi,

sorry for being so late but another project needs some working-time.

I have modified the code and will upload the new version here (maybe sunday). Here are some comments:

- in main.cpp I added the call to sdcard_init() and sds011_init()
- in cyclic.cpp function doHousekeeping() I added the sleep/wakeup-hanlding of the sensor, also the reading of the current PM-values 
- in senddata.cpp function SendPayload() I write the data to the SD-card and add PM10, PM25 to the buffer

Known problems:
- UART #2 of the ESP32 is used for SDS011 and IF482
- how is the format of the data send via LoRaWAN in the case of type 2,3,4

I will now look at my code and do some clean-up and will then upload it.

AugustQu, 2020-02-03:
the pull-request is here.

Can you please check the code.

One hint: in the hal-file you are using please add this line:
`#define HAS_SDS011  1      // we use the SDS011 connected to this board
`

AugustQu, 2020-02-05:
ah, these nasty copy+paste-erros -> corrected. 

The source should compile now except for the modifications needed for supprting the new display.

AugustQu, 2020-02-05:
ups, I forgot this library: EspSoftwareSerial

added.

cyberman54, 2020-02-05:
Why do we need EspSoftwareSerial?
ESP32 has two hardware UARTS, so let's try to keep cpu free from bitbanging?

AugustQu, 2020-02-06:
> Why do we need EspSoftwareSerial?

The SDS-software supports Hardware- and Software-serials. For this part the EspSoftwareSerial is needed.

2 possible ways to handle this:
- in SDS011 remove all references to SoftwareSerial
- use EspSoftwareSerial

both solutions have disadvantages.

cyberman54, 2020-02-06:
@AugustQu 

- I removed the library files from this repository due to unclear licence situation. But added the library in platformio.ini, what makes it automatically downloaded from github by the user while compiling the code.
- I also removed EspSoftwareSerial, since i do not see why we need it. We use UART#2 of esp32, thus we have a hardware serial, and as far as i see this should work with the SPS011 lib; at least the code does compile without error. Can you test current code in branch SDS011 on your device, please?

Perhaps we should look for another SDS011 library, which is better maintained and documented, and comes with a clear license.

AugustQu, 2020-02-06:
@cyberman54 
this is th error-message I got when I removed the lib  EspSoftwareSerial:
` *** [.pio/build/usb/src/main.cpp.o] Error 1
In file included from include/sds011read.h:4:0,
                 from include/payload.h:7,
                 from include/globals.h:135,
                 from include/irqhandler.h:15,
                 from src/irqhandler.cpp:1:
lib/SDS011/src/SDS011.h:17:28: fatal error: SoftwareSerial.h: No such file or directory
`

AugustQu, 2020-02-06:
this is my first project using PlatformIO so I do not know all the possibilities of this tool.

Downloading the source-code during the compile-time is a better approach. But I had to modify the file sds011read.cpp lin 26:
`//    sdsSensor.contmode(0);              // for safety: no wakeup/sleep by the sensor
`
With this modification the code compiles. 

But in the linking stage I got a lot of error.messages regarding the libBosch_BSEC.

AugustQu, 2020-02-06:
OK, the code compiles.

You made some modifications to the code so checking my code didn't work. So I downloaded a fresh copy of the SDS011-branch from here. This version compiled.

I will now check if it is working.

cyberman54, 2020-02-06:
sdsSensor.contmode() is not a member of the class in original SDS011 lib on github. It seems the lib is not maintained any longer, and there are several subtile versions around the net. Maybe better switch to a maintained lib. Look in platformio Home -> libraries -> search for SDS011

AugustQu, 2020-02-06:
oh yes. Another reason is the code produces a trap:

`[I][main.cpp:332] setup(): init fine-dust-sensor
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008227b  PS      : 0x00050034  A0      : 0x400856bc  A1      : 0x3ffbe7c0  
A2      : 0x3ffc1104  A3      : 0x00000002  A4      : 0x00000001  A5      : 0x400916ec  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ff6e000  A9      : 0x00000008  
A10     : 0x3ffbe7f8  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffcda40  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
Core 1 was running in ISR context:
EPC1    : 0x400e7771  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008227b

Backtrace: 0x4008227b:0x3ffbe7c0 0x400856b9:0x3ffbe7f0 0x400e776e:0x3ffcdaa0 0x400e77d1:0x3ffcdac0 0x400e78dc:0x3ffcdae0 0x400e4065:0x3ffcdb00 0x400e40a1:0x3ffcdb20 0x400e08ed:0x3ffcdb60 0x400d7ded:0x3ffcdb90 0x400d4f77:0x3ffcdbc0 0x400e7f3b:0x3ffcdc60 0x4008e909:0x3ffcdc80

Core 0 register dump:
PC      : 0x40182d9e  PS      : 0x00060a34  A0      : 0x800eaf85  A1      : 0x3ffbc1e0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800e9f5e  A9      : 0x3ffbc1b0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008eb1c  A13     : 0x3ffbc0e0  
A14     : 0x00000000  A15     : 0x3ffbbee0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40182d9e:0x3ffbc1e0 0x400eaf82:0x3ffbc200 0x400901b9:0x3ffbc220 0x4008e909:0x3ffbc240

Rebooting...
`

This is the reason why I added the SDS-library because the fix is this:
`void SDS011::begin(HardwareSerial* serial) {
  Serial.println("SDS011::begin");
//  serial->begin(9600);      do not call this function again
  sds_data = serial;
}
`
I will look for another library. 

PS: this lib worked without any modification on my Arduino-project.

cyberman54, 2020-02-06:
You could try to make a pull request in the [original repo of SDS011](https://github.com/ricki-z/SDS011) to fix the bug. 

I want to keep the paxcounter code clean of locally imported / patched libraries, to avoid license issues and to keep the dependencies maintained.

AugustQu, 2020-02-15:
in the last days I played with some libs for the SDS011. The one I like most is the one by Rajko from luftdaten.info. I have seen there is a new version here on Github and with this version my problems are solved (I have to modify my code a little bit).

PlafromIO still contains the old version. So how is a new version of a lib transferred to the PlatformIO-repository?

cyberman54, 2020-02-15:
Look at description of library registry in platformio.docs. The repo must contain. manifest file, e.g. library.properties or library.json. And a github release or tag with matching version number must be set. Then the platformio crawler will reveal the lib automatically. You also can register a lib manually by command: pio lib register <url>

cyberman54, 2020-02-15:
What about this one?

SDS011 sensor Library
#1563

AugustQu, 2020-02-17:
same problem as before.

What I need is this code:
`void SDS011::begin(HardwareSerial* serial, int8_t rxPin, int8_t txPin) {
	serial->begin(9600, SERIAL_8N1, rxPin, txPin);
	sds_data = serial;
}
`

This function is not included in the code you mentioned. But you can find it here:
[https://github.com/ricki-z/SDS011](url)

cyberman54, 2020-02-17:
@AugustQu It seems the SDS011 lib maintainer missed to edit the github tag / release version, which still says 0.0.5, but `library.properties` says 0.0.6.

Try to point platformio explicitly to the current repo version by adding this line to `platformio.ini` in the `lib_deps_sensors =` section:

https://github.com/ricki-z/SDS011.git#33fd8b6

I did this for you in the SDS011 branch. So you may try again compiling the code.

AugustQu, 2020-02-18:
OK, works.

In PlatformIO.ini I added:
`lib_deps_sds =
    https://github.com/ricki-z/SDS011.git#33fd8b6
`
and in lib_deps_all I added this line:
`   ${common.lib_deps_sds}
`
So it compiles.

I will do some tests now, clean my code and then upload my version.

Thanks for your help.

cyberman54, 2020-02-18:
The library maintainer must set correct release version in github. Please let him know by opening an issue, thanks.

AugustQu, 2020-02-19:
I made a pull-request from the SDS011-branch here to my SDS011-branch and added my modifications. I download a copy of this branch to my PC here and tried to compile the code. This is what I get:
`src/display.cpp: In function 'void init_display(bool)':
src/display.cpp:81:56: error: too few arguments to function 'int oledInit(int, int, int, int, int, int, int32_t)'
     oledInit(OLED_128x64, false, false, -1, -1, 400000L);
                                                        ^
In file included from src/display.cpp:36:0:
.pio/libdeps/usb/ss_oled_ID6458/src/ss_oled.h:58:5: note: declared here
 int oledInit(int iType, int bFlip, int bInvert, int iSDAPin, int iSCLPin, int iResetPin, int32_t iSpeed);
     ^
*** [.pio/build/usb/src/display.cpp.o] Error 1
`

Something has changed here. Can you please look into this.

cyberman54, 2020-02-19:
oled lib api and version have changed. Make sure the SDS011 branch is even with master, and you have a recent copy of ss_oled library in you local platformio (run "clean" before build). You need v3.3.1, not v3.3.0.

AugustQu, 2020-02-24:
sorry for the delay but I had some trouble with my internet-connection plus additional trouble here in the house.

cyberman54, 2020-02-25:
@AugustQu i made some code sanitizations. Could you please re-test?

Things still to be done:
- encode SDS values in Cayenne payload encoders
- add SDS decoding functions in TTN payload decoders / converters

AugustQu, 2020-02-26:
these are the open topics as I see them:
- add some lines to the Readme-file
- plus the ones you mentioned

I did not look into the payload-encoders, my current solution is a quick shot. So what I did is:
PAYLOAD_ENCODER-value 2 (packed): send the 2 values as integers in the form 10*pm10. As the range of this value is >=0 up to <2000 this causes no problems (same for pm25: >=0 up to < 1000)

PAYLOAD_ENCODER-value 1 (plain): I simply converted it into a string with a format %5.1f. Is this correct?

PAYLOAD_ENCODER-values 3 or 4: I did not look into the code until now. I will do this (or do you have a description of the format somewhere?).

AugustQu, 2020-02-26:
the code compiles and it seems to work.

AugustQu, 2020-02-26:
do I understand this correctly:

for  PAYLOAD_ENCODER-value 3 and 4 I need to add lines like these (in pseudocode):
 ```
  buffer[cursor++] = LPP_xxxxxxxxx;
    buffer[cursor++] = highByte(pm10-value);
    buffer[cursor++] = lowByte(pm10-valuevalue);
``` 

Is this correct? But where do I get the value for LPP_xxxxxx ?

AugustQu, 2020-02-26:
can I say:
`#define LPP_PARTMATTER_CHANNEL 32
#define LPP_PATMATTER 1xx
`
and write the code using these?

I did a quick look into the docs and did not found any hints that data for particulate matter is supported, but I found a hint that they only support the defined types.

What to do here?

cyberman54, 2020-02-26:
You need to decide which Cayenne sensor you hitchhike for transmitting PM values, since Cayenne does not know PM sensors, and there is no option to have user defined sensors in Cayenne protocol. Since this protocol wasn't touched as of 2019, i assume it looses relevance; but they have a nice app for visualization.

Look here for description:

https://github.com/myDevicesIoT/cayenne-docs/blob/master/docs/LORA.md
https://community.mydevices.com/t/cayenne-lpp-2-0/7510/17

AugustQu, 2020-02-27:
I like the temperature-example. So I can transmit the PM-values in the form (in hex):
`03 67 00 9D`
for 0x9D -> 157 -> 15.7 for PM10. And so
`05 67 00 19`
for 0x19 -> 25 -> 2.5 for PM25.

Is this OK?

I found the temperature-example in the first link you mentioned.

AugustQu, 2020-02-27:
to me the code can look like this:
`#if (PAYLOAD_ENCODER == 3)
    buffer[cursor++] = LPP_COUNT_WIFI_CHANNEL;
#endif
    buffer[cursor++] =
        LPP_LUMINOSITY; // workaround since cayenne has no data type meter
    buffer[cursor++] = highByte(value);
    buffer[cursor++] = lowByte(value);
`
naturally with different values.

This looks easy after we fixed the constants.

AugustQu, 2020-02-28:
I've added some code for handling the Cayenne-format.

Can you please check this.

AugustQu, 2020-02-28:
I will be offline for some days now so I cannot come back to this discussion until the 9th of march.

AugustQu, 2020-03-09:
I'm back in town and continued my tests.

With a setting of _PAYLOAD_ENCODER 3_ I can see this in the TTN-console:

> 15 65 00 03 20 65 00 00 21 65 00 00
15 65 00 03 20 65 00 00 21 65 00 00
15 65 00 05 20 65 00 00 21 65 00 00

With the following interpretation:
15 LPP_COUNT_WIFI_CHANNEL
65 LPP_LUMINOSITY
00 03 number of smartphones found
20 LPP_PARTMATTER10_CHANNEL (I added this)
65 as above
00 00 the value of pm10 (it's 0 because of a fault on my side)
21 LPP_PARTMATTER25_CHANNEL (I added this)LPP_PARTMATTER25_CHANNEL
65 as above
00 00 as above

For me this looks good in the TTN-console.

By setting _PAYLOAD_ENCODER 4_ I receive a lot of compile-errors outside of my code like:

> src/payload.cpp: In member function 'void PayloadConvert::addCount(uint16_t, uint8_t)':
src/payload.cpp:358:9: error: 'LPP_LUMINOSITY' was not declared in this scope
         LPP_LUMINOSITY; // workaround since cayenne has no data type meter
         ^
src/payload.cpp: In member function 'void PayloadConvert::addAlarm(int8_t, uint8_t)':
src/payload.cpp:378:22: error: 'LPP_PRESENCE' was not declared in this scope
   buffer[cursor++] = LPP_PRESENCE;
                      ^

So it looks like this part of the code has not been used before.

cyberman54, 2020-03-17:
Thanks to the work of @AugustQu this feature is now implemented, and integrated in development branch. Will be merged with master with next release.

closed 2020-03-17 by cyberman54