cyberman54/ESP32-Paxcounter #271: issue with build in time.cpp

TizianCedalo, 2019-02-12:
While compiling with latest version we get this problem and it aborts building

> 
*******************************************************************

compilation terminated.
Compiling .pioenvs\ttgov21new\lib532\Time_ID44\Time.cpp.o
*** [.pioenvs\ttgov21new\src\rtctime.cpp.o] Error 1
Compiling .pioenvs\ttgov21new\lib520\Timezone_ID76\Timezone.cpp.o
======================================================================== [ERROR] Took 12.98 seconds ========================================================================

>

cyberman54, 2019-02-12:
I currently cannot reproduce this, wondering what "Error 1" is.
Please try
- use platformio Clean function before build
- try current version of development branch

If the error persists, please try to gather more context information and post it here.

TizianCedalo, 2019-02-14:
Thank you the new version seemed to have fixed it!

closed 2019-02-14 by TizianCedalo