cyberman54/ESP32-Paxcounter #276: ttgo beam with sds 011

sbamueller, 2019-02-20:
my ttgo beam does not transmit any gps data.
what could be the cause?

[image]

cyberman54, 2019-02-20:
Probably one of these reasons:

- Your GPS has no fix, thus has not yet sent any NMEA record with valid(!) location data
- You did not configure GPS or you set erraneous parameters for the GPS interface (serial params / i2c addr) -> in <board>.h file
- GPS interface hardware not working -> check wiring
- GPS data transmit was switched off before and saved to NVRAM, thus reloaded at reboot -> rcommand 0x03
- You have setup an active payload decoder/converter in TTN console which filters the data

cyberman54, 2019-02-20:
Ok, got it: with one of last commits GPS was disabled by default in `ttgobeam.h` file, i changed it. Please update ttgobeam.h and reflash.

closed 2019-02-20 by cyberman54