cyberman54/ESP32-Paxcounter #280: Store packages in microSD if not able to send them

davidex720, 2019-02-24:
Using tbeam it could be really helpful to have the opportunity to store messages that were not successfully sent.
Doing so it could be possible to make something like a buffer of older message that the device could send once it reconnects to a gateway. 
Using GPS lock the tbeam could store the messages with both time and gps coordinate

cyberman54, 2019-02-24:
You're welcome to make pull request to implement this, shouldn't be too complicated since we have all necessary parts already in the code:

- There is already a queue management implemented in `senddata.cpp` which buffers all outgoing messages. This makes it easy to implement a bigger buffer. Just increase size of RTOS queue and store it whereever. T-Beam has 4 MB PSRAM, this should be sufficienct for thousands of messages, no need for microSD)
- Iead of issues reading data from SD card during runtime to application; not sure if SD cards on those boards can be used for other purpose that boot the device from flash image
- concerning time stamping of messages: in development branch you find an enhanced version with `timekeeper.cpp` which tries to keep system time synched with an external timesource, e.g. GPS. Using this version you can just take systime to timestamp messages.

davidex720, 2019-02-24:
very cool!
the idea to use a microSD is coming from the need to validate the package sniffing, a lot easier on a microSD than on PSRAM
very cool the gps time implementation, I'll have a look at that

cyberman54, 2019-02-24:
What do you mean by _validate the package sniffing_ ?
The intention of paxcounter is to have a privacy respecting simple counter, not to have an offline wireshark device for collecting sniffed data.

davidex720, 2019-02-24:
i was thinking about tests

cyberman54, 2019-02-24:
>  a lot easier on a microSD than on PSRAM

I don't see that it makes a big difference how and where to store the data.

@davidex720 You just need to rewrite the [mallocator function](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/include/mallocator.h) to store data where and how you want it, e.g. on a filesystem.

davidex720, 2019-02-24:
ok, thank you, i'll do it

closed 2019-02-24 by davidex720

cyberman54, 2019-02-24:
@DavideNesi If you do it, would be nice to see your pull request here after.

cyberman54, 2019-03-08:
@davidex720 for SD card usage with ESP32, see here:

https://github.com/espressif/arduino-esp32/issues/1190#issuecomment-470721024