cyberman54/ESP32-Paxcounter #293: Output DEVEUI to serial console as soon as VERBOSE is set to 1

gregorwolf, 2019-03-06:
The Readme currently states:

> The DEVEUI for OTAA will be derived from the device's MAC address during device startup and is shown as well on the device's display (if it has one) as on the serial console for copying it to your LoRaWAN network server settings.

But currently it's not enough to set 

```
// Verbose enables serial output
#define VERBOSE                         1
``` 

in paxcounter.conf. Also the `debug_level` must be set to at least 3 in platformio.ini.

I would suggest to either mention that in the readme or change the behaviour so that the DEVEUI is sent to the serial console as soon VERBOSE is set to 1.

cyberman54, 2019-03-06:
Intention of verbose is to completely mute a device, since serial output could cause trouble on production devices. Thus, i will change it that way, that DEVEUI is always printed on console if verbose is set to 1, regardless of debug level

cyberman54, 2019-03-07:
added to readme.md

closed 2019-03-07 by cyberman54